MarlinFirmware/Marlin #13688: [bugfix-2.0.x] too many #define DEBUG_OUT ENABLED(xxx) in a lot of files which causing DEBUG_ECHO_XXXX function only "NOOP"

scott0122, 2019-04-14:
### Description
For example:
In  configuration_store.cpp line 401
  #define DEBUG_OUT ENABLED(EEPROM_CHITCHAT)

But when I enable AUTO_BED_LEVELING_UBL without DEBUG_LEVELING_FEATURE in Configuration.h, DEBUG_OUT is turned off in ubl.h( #define DEBUG_OUT ENABLED(DEBUG_LEVELING_FEATURE) )

This cause configuration_store's _load() and reset() response nothing.

[image]

ManuelMcLure, 2019-04-14:
This is probably related to #13641 - I wonder if removing the
```
#pragma once
```
from ``src/core/debug_out.h`` would help.

scott0122, 2019-04-15:
My solution is add 2 line to src/core/debug_out.h

#undef DEBUG_OUT
#define DEBUG_OUT ANY(DEBUG_LEVELING_FEATURE, DEBUG_POWER_LOSS_RECOVERY, MMU2_DEBUG, DEBUG_DXC_MODE, L6470_CHITCHAT, EEPROM_CHITCHAT)

ManuelMcLure, 2019-04-15:
Maybe removing the
```
#pragma once
```
and instead of directly setting
```
#define DEBUG_OUT ENABLED(<variable>)
```
define a ``DEBUG_ENABLE(<variable>)`` macro like this:
```
#define DEBUG_ENABLE(x) \
   #if !defined(DEBUG_OUT) || !(DEBUG_OUT) \
       #define DEBUG_OUT ENABLED(x) \
   #endif 
```

ManuelMcLure, 2019-04-15:
On the other hand, an easy fix for the ``EEPROM_CHITCHAT`` is to move the following two lines in ``configuration_store.cpp``:
```
 386 #define DEBUG_OUT ENABLED(EEPROM_CHITCHAT)
 387 #include "../core/debug_out.h"
```
up above these lines:
```
  63 #if HAS_LEVELING
  64   #include "../feature/bedlevel/bedlevel.h"
  65 #endif
```

ManuelMcLure, 2019-04-15:
From code examination, it looks like the only header file that calls "DEBUG_OUT" is ``ubl.h``, so it's probably sufficient to remove the
```
#pragma once
```
and ensure that any .cpp files that depend on ``DEBUG_LEVELING_FEATURE`` include ``debug_out.h`` themselves (it looks like they already do).

ManuelMcLure, 2019-04-16:
Try pull request #13716 - that should fix the problem.

closed 2019-04-16 by thinkyhead