MarlinFirmware/Marlin #13693: [BUG][bugfix 2.x] Bed heating bad behaviour

GMagician, 2019-04-14:
This follow my previous #13070 and #12061.

I'm back with this because today I had same issue...usually I can work with bed temp of 90° (its slow to reach it but it can keep 90° for a whole 1h print).
Sometime (rare) I face a complete lock in temperature control. I faced always on output and out always off output with a correct thermistore feedback.
I'm wondering if this is not related to my "slow" heating that lead to an overflow/underflow computation.

My suspect are on:
`if (pid_error > 0) temp_iState -= pid_error; // conditional un-integration`

if pwm stays too long to 100% temp_iState may overflow.

I'm sure that HW works perfectly because when problem is present I may solve it turning on/off bed heating using "repetier button". In one situation (bed was always on) turning off and on a couple of time solved for the whole print. This is why I think about some overflow computation

7eggert, 2019-04-15:
I'm using limit switching with a hysteresis of 0.2 °C - this might be a temporary solution for you. HTH

GMagician, 2019-04-16:
I'm investigating on it...Not sure but I think I saw a target set point to 0...but that's not in gcode...

GMagician, 2019-04-30:
I found a bad cable...not sure about it but I close this till next event...

closed 2019-04-30 by GMagician