MarlinFirmware/Marlin #13695: [BUG][1.1.x-bugfix] second printhead crashes on IDEX-Machine in DXC_FULL_CONTROLL_MODE

cad435, 2019-04-14:
### Description

If running a dual X-Carriage/IDEX setup, after the Second Tool-Change commands, the inactive extruder position got lost (I checked it and it is set to X=0.0, it should be X=235.5).
At the nex G1 command wich adresses said X2 Carriage, it slamms into the end, because it thinks it is at X=0.0 and therefore moves positive direction, when infact it should move negative direction.

This seems to happen only if HOTEND_OFFSET_X/Y/Z of the second carriage is not set zero.

### Steps to Reproduce

- set Z-Offset to 0.5 (either in source File or via a M218 command)
- make a T1 command, followed by a T0 comman, followed by a T1 command.
- check your X-Axis position --> it should be your X2 axis position but it reports X=0.0

**Expected behavior:**

second T1 command should restore the X2 position and therefore should not slam into the X2-Endstop

**Actual behavior:** [What actually happens]
after second T1 command T1/X2 X-Pos is set to 0.0

#### Additional Information

Configuration.h https://pastebin.com/RnrH7VHD
Configuration_adv.h https://pastebin.com/u6imKSSc
Pins_RAMPS.h https://pastebin.com/4Ygjhht4

I wrote a little Bugfix-Hack wich is working for me, as it seems the codeline 
"inactive_extruder_x_pos = destination[X_AXIS];" (Marlin_main.cpp, Line 12280) is not working an will store the 0.0 into inactive_extruder_x_pos 

you can find the modified piece of code here: https://pastebin.com/kqXXH2TE or on this commit from my forked Marlin Repo:
https://github.com/cad435/Marlin/commit/7b41d20be1104ca79013fa5287171194d6bc15e3

This is possible not the best fix, as it is just programmed around...

For my setup it is right now working perfectly fine and it'll maybe help you (if theres any real Bug and I have not just messed up my config in a way I'm not aware of...)

cad435

InsanityAutomation, 2019-04-14:
At least 3 methods of this occurring have been fixed in Marlin 2.0 already. Can you test with bugfix-2.0.x and verify if you can reproduce?

This looks like one of the ones fixed within weeks of when 1.1.x was finalized to 1.1.9 and all future work was 2.0.x focused.

cad435, 2019-04-14:
The Last time I tried to configure Marlin 2.0 I ended up with functionla FW but without a functional screen, but I'll try again and Post here If i succeed...

cad435, 2019-04-14:
All right, I got Marlin 2.0-bugfix running wich is... lets say "different" than 1.1x-bugfix

DXC_FULL_CONTROL_MODE don't work like expected: it should do nothing. Instead it raises Z-Axis, and tries to move the new carriage to the position of the old one (like unpark from DXC_AUTO_PARK_MODE). No crashes at this point, X1 is in Home, X2 tried to move to X=0, but will stop on X=30 because of the '#define X2_MIN_POS    30'

But on next toolchange from T1/X2 to T0/X1 it won't remeber to park T1/X2 (and actually behaves like DXC_FULL_CONTROL_MODE ), so it basically slamms one carriage into another.

This is not tested with a gcode file, only from LCD "Switch to E1/E2", I'd rather not ruin my printer on constant carriage crashes...

Well, Z-offset seems to be working at least somehow like expected, but I'll and can only test it further if I'm able to controll theese carriages from the LCD correctly.

Again, my config-files for 2.0-bugfix
Configuration.h https://pastebin.com/VTvNXUbF
Configuratio_adv.h https://pastebin.com/w7qsf1U4

I did NOT tinker around with any sourcecode this time, "stock" firmware here...

cad435

InsanityAutomation, 2019-04-15:
Ok, should be enough to go on for some debugging on this end. Auto park has been made the default as full control mode has been much less tested as of late. Is there a specific need for full control mode that auto park does not fulfill or does that work as a temporary solution?

cad435, 2019-04-15:
There is a specific nedd, I would say:

lets say we print with ABS on T0(X1) and PVA (water-soluble support material) on T1(X2). This is needed if you have for example complex parts wich must withstand a bit of strength, or Parts wich have overhangs wich needs to be precise.

So, lets say we are at the end of our ABS part from a Layer and our Nozzle is at X=10/Y=10. Now we request a Toolchange, T0 would raise and park. T1 would unpark, move to X=10/Y=10, lower on that coordinates, and AFTER lowering get the next line of g-code and move to the first coordinates of the PVA Part of that layer. Wich will leave a bit of PVA on the ABS Layer. Wich will decrease strenght or can even mess up whole parts.

same thing applys when printing with clear materials, as you would end up with a bit of opaque color in your transparent Parts.

Above is tested only wich 1.1.x-bugfix, I assume the same applys for 2.0, pls correct me if I'm wrong...
Nevertheless, I'd think full-controll mode should be working, as controlling Toolchange parameters from a slicer is more convinient other than reflashing the Firmware every time you make changes

cad435

InsanityAutomation, 2019-04-15:
> Above is tested only wich 1.1.x-bugfix, I assume the same applys for 2.0, pls correct me if I'm wrong...
Yup, same applies. Weve discussed a look ahead for next position command for the return but that hasnt been done yet. Would need to determine if E move is present or not and if so return to original position. If not go to that command.

> Nevertheless, I'd think full-controll mode should be working, as controlling Toolchange parameters from a slicer is more convinient other than reflashing the Firmware every time you make changes
Agreed, and why I said temporary solution not long term! Thisll get worked on soon for sure.

cad435, 2019-04-15:
> Thisll get worked on soon for sure

Very nice! I'll test it for sure if theese fixes are made

cad435

InsanityAutomation, 2019-04-23:
Please check if the current snapshot resolves the issue on your end. If so, please close this issue.

Thanks

cad435, 2019-04-23:
It indeed seems fixed now. X-Axis displays right coordinates after changes now and won't move anymore. Thank you for fixing it!

Two things wich didn't work like I would expect them to work (but I'm not sure if you guys think the same)

First: after pressing "Switch to E1/E2" the LCD goes back to the status screen, wich is kind of annoying.

second: Z-offset should be corrected as soon as the Tool-change is requested. In most gcode is no "G1 Z" command after Toolchange, so the Printer knows now it is offset but won't correct the offset till the next Layer-change, as there is a absolute "G1 Z" command. Same thing applies to all axis if your gcode contains incremental movements (but personally I don't know anyone who does this, nor do I know if theres any benefits)
In Slic3r I can put a 'G1 Z[layer_z]' command after my Tool-Change G-code, wich will move the Z-Axis and therefore actually correct the Z-Offset. I don't know if other slicers have such a "Z-Layerheight" Variable.

I'll close this issue now because the original movement issue is fixed now.

cad435

closed 2019-04-23 by cad435

InsanityAutomation, 2019-04-23:
For the first issue, it was a debate based on safety. If youre in a babystepping or jog motion screen theres a potential of movement to an unintended axis. The alternative is exposing active screen globally, and tracking safe / not safe screens for multiple scenarios.... Not really maintainable. Im all ears for a better action to take  if someone can find one though!

For the second i think I agree. Can either re-open this with that added to the OP, or make a new issue and tag me in it and ill keep it on my list to expand on a little further. Its a bit less pressing than the original issue here but still needs some attention at some point.