MarlinFirmware/Marlin #13696: UI steps adjustment error

cinealfa, 2019-04-14:
Steps adjustment for extruder doesn't work as expected

1 Go to steps per mm menu
2 try to adjust extruder steps let's say the current setting is 400 steps
.

**Expected behavior:** [What you expect to happen]
The number should reflect the setting in memory and you should be able to go higher than 168.

**Actual behavior:** When the adjustment number is brought up it is now 5
 If you try to bring the adjustment back to 400 it will not go higher than 168.
 if you stop at 168 and the reselect it will NOT go any higher. You can reset the eeprom and the correct number will return
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/3077591/Archive.zip)

zombu2, 2019-04-14:
i see the same thing happening also extrusion does not work when you use the display and it resets the board

cinealfa, 2019-04-15:
interesting, What board are you using? I am using MKS Sbase

zombu2, 2019-04-15:
bigtreetech SKR v1.3

marcio-ao, 2019-04-15:
I confirm this problem on an 8-bit RAMBO board.

marcio-ao, 2019-04-15:
I know what is happening. Will make a PR with the fix.

marcio-ao, 2019-04-15:
So, the reason this is happening is because the format used for steps is float62. This has a multiplier of 100:

```
DECLARE_MENU_EDIT_TYPE(float,    float62,     ftostr62rj,    100     );   // 1234.56    right-justified
```

At  800 steps, 800*100 is 80,000, which is greater than the maximum int16_t values used in MenuItemBase:

```
class MenuItemBase {
  private:
    ...
    static int16_t minEditValue, maxEditValue;
    ...
}
```

Notice that the maximum values expressed by `int16_t` is 0x3FFF, which translates to 16383. Divide out the multiplier of 100 and you get 163.83.

So, a couple possible solutions:
  1) Change the types of minEditValue, maxEditValue to int32_t
  2) Change the formatting used in the menu items for steps from float62 (which uses a multiplier of 100) to float51 or float3 (which uses a multiplier of 10 and 1, respectively)

In our LULZBOT FW, we use float3, because it does not make sense to us to have faction of steps. If there is a need for maintaining fractional steps, changing the format to float51 will allow a 10th of step while increasing the range from -163.83 to 163.82 to -1638.3 to 1638.2, which is probably sufficient for most cases.

@thinkyhead: I was going to do a PR, but since I am not sure what solution you prefer, I am throwing this in your court :)

marcio-ao, 2019-04-15:
Actually, I just tried float51 and it allows the value to go negative... So that's not a fix.

cinealfa, 2019-04-16:
Marcio,
What a clear concise explanation. Thank you very much.
Does it make sense to put some check to prevent a negarive number?

marcio-ao, 2019-04-16:
> Given the pervasiveness of the problem, I think changing the types of minEditValue and maxEditValue from int16_t to int32_t. The downside of this is that it may increase Marlin's compiled size by quite a bit.

Just an updated, this didn't work either. The fundamental problem is that the encoder using an int16_t. See my PR for an alternative fix that lops one decimal point of the steps per mm adjustment in order to give an effective range of 0.0 through 3276.7

marcio-ao, 2019-04-17:
@cinealfa: Can you check the latest build and close this ticket if it resolves the issue for you?

doggyfan, 2019-04-17:
Still faulty for me in the latest build.

marcio-ao, 2019-04-17:
doggyfan: What is your steps/mm set to?

doggyfan, 2019-04-17:
X: 40
Y: 40
Z: 400
E0: 84.9
 
I go to edit them on the LCD I get 3108.2 with all of them.

marcio-ao, 2019-04-17:
Hummm, interesting. Let me double check.

marcio-ao, 2019-04-17:
I'm definitely not seeing that problem. Do you have an 8-bit or 32-bit board?

doggyfan, 2019-04-17:
32-bit BTT SKR v1.3

doggyfan, 2019-04-17:
It's OK on the 8-bit MKS board, just not on the BTT SKR 32-bit board,

marcio-ao, 2019-04-17:
@doggyfan: Okay, I'll rig up a 32-bit board and do some testing.

marcio-ao, 2019-04-17:
@doggyfan: Able to confirm this on a 32-bit Archim. I will look into it. Thanks for reporting.

marcio-ao, 2019-04-17:
@doggyfan: Easy fix. I made a PR for it. #13737

doggyfan, 2019-04-17:
Well done :)

I'll test it when it's out.

doggyfan, 2019-04-17:
That's fixed it :)

Well done and thank you Marcio.

cinealfa, 2019-04-20:
I will try the fix as well thank you @marcio-ao

cinealfa, 2019-04-20:
Well I cant seem to compile I dont recal changing anything but 
[image]
now I get this error: 
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/3100558/Archive.zip)

cinealfa, 2019-04-22:
still unable to compile...

cinealfa, 2019-04-26:
anyone?

marcio-ao, 2019-04-26:
@cinealfa: No clue. We build both 32-bit and 8-bit firmwares and I have not come across that error. Have you done a pull recently from the repo?

cinealfa, 2019-04-26:
marcio-ao thank you for your response
Forgive me but I don't know what "a pull recently from the repo" means. I am a bit of a noob :)

marcio-ao, 2019-04-26:
@cinealfa: Try the absolute latest source.

cinealfa, 2019-04-27:
Well good news, bad news I now can successfully compile the latest version but unfortunately I get the thermal runaway error #13783 so I am stuck until that is actually addressed. Frowny face lol

cinealfa, 2019-04-27:
Works as advertised!!!!

closed 2019-04-27 by cinealfa

marcio-ao, 2019-04-29:
@cinealfa: Glad to hear!