MarlinFirmware/Marlin #13697: [2.0] Solution SD card not recognized (working)

roel8032, 2019-04-14:
Please Developers, put it in next bugfix release.

In my case for Anet A6 and Infitary M508.

In file: main.cpp
In directory: C:Marlin\src
Find: void setup() {

Place here under:
#if ENABLED(SDSUPPORT) // Automatic recognize SD Card
       card.initsd();
     #endif

So it will be:
void setup() {

#if ENABLED(SDSUPPORT) // Automatic recognize SD Card
       card.initsd();
     #endif

and so on.

gloomyandy, 2019-04-14:
Unfortunately it is not as simple as it might first seem. Some users do not want the card to be mounted automatically (because on some 32 bit boards the SD card is shared via USB with a PC and they prefer to have the card available to the PC on boot), Then there is the relationship between the SD card and the LCD code to be taken into account, plus what should be done about running the auto-start code. Oh and things may be different if you have an SD card detect pin. So if anyone is planning to submit a PR you may want to read...
https://github.com/MarlinFirmware/Marlin/pull/13288
and
https://github.com/MarlinFirmware/Marlin/issues/13274

Personally I'm still not really sure what the best solution is. I think I'd really like to see an option that allows the user to choose to either mount the card on boot (and then do any associated auto run stuff) or not. This would be the case with/without an LCD and with/without an SD detect pin. However this would change the timing of when the SD card gets initialised for a large number of users (currently it will be done as part of the LCD periodic timer), which may have some unexpected consequences.

7eggert, 2019-04-15:
Compile time default (not) to mount the card for Marlin or a variable in EEPROM would be my favorites.

If the LCD timer can mount the card, there will be a variable telling it (not) to do this. This may be set to a default? I did not look at the code yet.

ArturNadolski, 2019-04-15:
Maybe solution is to add 2 flags for enabling/disabling SD mount? One for onboard SD and second for SD with LCD?

7eggert, 2019-04-15:
The code I remember will just mount any SD card, searching the onboard slot first. That would be a major change (not that I', opposed, JFYI).

gloomyandy, 2019-04-15:
Marlin can currently only support a single SD card, other configuration that already exists specifies if this is the LCD SD card or the onboard one (if it exists). So I think only a single new configuration option would be required (to specify if the card should be mounted on boot), with the default probably being to mount it. I suspect that the code to mount the card from the LCD timer is not really needed as Marlin seems to be happy to mount the card from the main startup code in the case of no display. But we will probably not really know until the change is made.

7eggert, 2019-04-15:
You can have SDSS and LCD_DSSS defined; then it will do:
```
  if (!sd2card.init(SPI_SPEED, SDSS)
    #if defined(LCD_SDSS) && (LCD_SDSS != SDSS)
      && !sd2card.init(SPI_SPEED, LCD_SDSS)
    #endif
  ) {
```
Thus you'll never know which SD card would be activated by `CardReader::initsd()`.

gloomyandy, 2019-04-15:
Yes you can have both SDSS and LCD_SDSS defined but Marlin will only access one SD card at a time (so I really don't think it makes sense to have two flags that would add to the confusion) and on systems with multiple readers it would make sense to ensure that either LCD_SDSS is not defined or it is equal to SDSS (so only a single reader will ever be used). I'm pretty sure this is not the case at the moment. 

What I mean by the LCD timer code that mounts the card not being needed is that it is not necessary to delay the mount of the SD card until the LCD timer is called (one of the things mentioned by thinkyhead in the previous discussion) as it seems to be fine to call the mount code during setup when an LCD is not provided. I think in the case of the LCD keyboard all that code does is to queue an M22 (or whatever) gcode request which then gets executed via the normal gcode execution loop (so not from the LCD timer code). But this stuff is all pretty complex so I may be wrong.

7eggert, 2019-04-15:
The intended behavior of SDSS != LCD_SDSS would be to support both SD slots even though only one SD card can be used, as some users will use this one and some will use that one.

I did not yet look at the code that would trigger the mount, but if it's done by gcode, then a printer start gcode would be a more generic way to trigger the intended behavior.

gloomyandy, 2019-04-15:
Well the menu code certainly queues a gcode request. But I'm not sure that doing the startup mount would work if you queued a gcode request. The problem is there is a lot of other code in the startup sequence that can end up checking for various files (like the recovery from power fail and execute gcode on start code), I'm not sure but I have a nasty feeling that delaying the mount until gcode is executed could mess up some of this code. As I said above it is not a simple change to make (and be sure that nothing gets broken).

As to automatically selecting which reader to use I don't think it is a good idea, but I suppose someone may want it. But even so with the LPC176X boards there is a strong case for being able to force Marlin to use the LCD reader (even if there is a card in the on board reader) because many users like to have the on board SD card there for firmware updates, but still use the LCD reader for gcode. At the moment if you just rely on the automatic selection I don't think this is possible.

7eggert, 2019-04-16:
Mounting at boot: I cannot try this as I don't use these features. I noticed a function to drain the gcode queue though; I think using that would suffice. Also the generic solution would have a better chance to be accepted.

SD card order:
Just exchange the #defines to prefer the LCD SD card reader. My ¢¢ for this side note, the LCD card reader should be examined first if it exists for exactly this reason.

gloomyandy, 2019-04-16:
I think we may just have to disagree on this. 

Going through loads of gcode queuing, decoding, draining etc. just to make a single function call  seems like overkill to me. At the end of the day the M21 gcode handler simply calls card.initsd(). 

As to trying first one card reader and then the other, well having thought about it a little more I've realised that for some important control boards it simply doesn't work!. The try one and then the other chip select lines only works if both SD card readers share an SPI bus. Guess what, on some boards (like the RAMPS ReArm, the Bigtreetech SKR V1.3 and possibly others) this is not the case. On these the two readers use different (hardware) buses. But the Marlin SD card code will always just use one of those buses. This also means that if LCD_SDSS != SDSS then the sd card code will end up setting chip select on a reader but then sending SPI signals out on a totally different bus, which is arguably a bug and almost certainly not a good thing to do. 

This means that we either need to fix the SD card reader code so that it works with different SPI instances (no bad thing, but a fair bit of work), or we make the user choose which reader will be used at configuration time (which is in effect what happens on the LPC176X based boards today). 

As I said in my first post, this stuff can be rather complicated and is certainly not easy to test!

7eggert, 2019-04-26:
Sorry for the delay. I agree that if calling one function unconditionally would be done better by just calling it:-), but this feature has been rejected before as it might be undesired. There would need to be a Cconfiguration.h-option, too ...

However if people feel the need to have a startup code (some do), having a ROM-based init string - or maybe for future versions an EEPROM based one - might be the better option: Have one feature to handle everything and be done.