MarlinFirmware/Marlin #13699: [FR] True filament sensing

klcjr89, 2019-04-14:
## Description

I have finished working on my true volumetric filament sensor, and I believe it requires the code in `filwidth.cpp` and `planner.cpp` to be changed (not sure about other files), as the old code assumes measurement is done in one axis.

My sensor measures the filament 360º around the periphery of the filament, and outputs an analog voltage that is a delta value. The sensor is calibrated with a precision ground 1.750mm OD pin gage, used as a standard.

Since my sensor reports the voltage based on the cross section and not one axis like the old filament sensors, I am not sure what formula needs changed in the source files.

For sanity, it may be best to have a macro that specifies whether the sensor is a single axis (width/diameter) sensor, or a true volumetric sensor, that way the user can specify which sensor they have when I begin selling the sensors.

doggyfan, 2019-04-15:
Curious as to what method you're using to measure the filament cross sectional area ?

klcjr89, 2019-04-20:
Any update on what formula needs changed and how this can be implemented for the main branch?

klcjr89, 2019-05-08:
@thinkyhead

thinkyhead, 2019-05-08:
The `FILAMENT_WIDTH_SENSOR` in Marlin reads a resistance via ADC and uses this to determine the likely width of the filament. This function converts the latest ADC reading to a millimeter value:
```cpp
  // Convert raw Filament Width to millimeters
  float Temperature::analog_to_mm_fil_width() {
    return current_raw_filwidth * 5.0f * (1.0f / 16383.0f);
  }
```

klcjr89, 2019-05-08:
@thinkyhead I realize that, but my sensor outputs a voltage based on the circumference of the filament. So I'm not sure how to figure out the formula.

lrpirlet, 2019-05-08:
circumference = Pi * Diameter…
Pi is a constant so, your function will end-up with a parallel function…
So, if you make sure that a circumference of 3*Pi outputs a voltage of 3 volts a circumference of 2.8*Pi will output 2.8 volts… 
KISS
:-)

lrpirlet, 2019-05-08:
Haha my text got formatted unexpectedly, please read 
circumference = Pi * Diameter…
Pi is a constant so, your function will end-up with a parallel function…
So, if you make sure that a circumference of 3 * Pi outputs a voltage of 3 volts a circumference of 2.8 * Pi will output 2.8 volts…
KISS
:-)

AnHardt, 2019-05-08:
If your sensor is measuring the filaments **circumference** instead of its **area** you don't get any advantage over measuring the **diameter**.
As long the filament is **round** non of the different measured values has an advantage over an other, it's just an other calculation to find the filaments area (what we are looking for). As soon as the filament is not round only directly measuring the **area** will give you the **area** (and i have no clue how that could work). 
Calculating the **area** from the **circumference** is impossible without a assumption about the shape of the measured object. Even if you assume an ellipse (the second simplest shape after the circle ) the area/circumference relation depends on the shape of the ellipse. (If the small and the big length of the axes are identical you get a circle. If the small axis goes to zero the circumference goes to two times the large axis and the area goes to zero.)

What is your sensor measuring?
How does it work?

AnHardt, 2019-05-08:
> when I begin selling the sensors.

Confusing *circumference* and *area* by the inventor/producer/seller of a sensor shrinks the trust into the product to below zero.

klcjr89, 2019-05-08:
Sorry to confuse you. Periphery may be a better term. I don't care if the filament isn't round or not, It could be square shaped and it wouldn't matter in my case.

What I need to know is what formula to change in Marlin, but that's after the filament sensor code is fixed to not cause my LPC1769 to freeze.

InsanityAutomation, 2019-05-14:
Honestly aside from the difference with a 3.3v analog input on the LPC1768/9 you may not need to. Youre looking at standard deviation from setpoint moreso than anything else. It may need to be done in the HAL per platform based on logic level. Ill have to take a look once i have something to hook up and look at now that.

sjasonsmith, 2020-11-10:
@klcjr89 is this still something you are working on?