MarlinFirmware/Marlin #13701: [bugfix-2.0.x] FILAMENT_WIDTH_SENSOR causes controller to hang

klcjr89, 2019-04-14:
### Description

Enabling the `FILAMENT_WIDTH_SENSOR` feature causes my Azteeg X5 mini v3 controller board to hang/freeze and is nonresponsive.

### Steps to Reproduce

• Uncomment the `FILAMENT_WIDTH_SENSOR` definition
• Compile firmware and place on SD card
• Turn power switch on to controller and let it flash the new firmware
• Firmware/controller freezes after Marlin boots after a few seconds has passed.

**Expected behavior:**

I expect the controller board to not freeze.

**Actual behavior:**

The controller board freezes and is unusable.

#### Additional Information

Zip file attached of config and board pin files.
[ConfigFiles.zip](https://github.com/MarlinFirmware/Marlin/files/3077929/ConfigFiles.zip)

psavva, 2019-04-15:
Do you get any output from the serial monitor via your host software?

klcjr89, 2019-04-15:
> Do you get any output from the serial monitor via your host software?

I don't use host software.

klcjr89, 2019-04-18:
Any update on fixing this?

Roxy-3D, 2019-04-18:
Using Host software will provide more information than what we have now.

Any update on the full set of error messages being produced?

klcjr89, 2019-04-18:
I don't see how host software will report anything if the controller freezes. Regardless, I will hook it up if you want me to.

klcjr89, 2019-04-18:
I just connected it to my computer, and it doesn't report anything of use before the controller freezes and disconnects from the computer..

[image]

klcjr89, 2019-04-18:
Do I need to send M111 before the controller hangs?

klcjr89, 2019-04-20:
Any update on how to fix this and debug?

klcjr89, 2019-04-30:
@Bob-the-Kuhn Any tips?

klcjr89, 2019-05-08:
@thinkyhead What can be done? I'm struggling to figure this out.

InsanityAutomation, 2019-05-14:
Discussed with the user on Discord and we found the issue. @klcjr89 can you close this once the above PR is merged?

klcjr89, 2019-05-14:
@InsanityAutomation Absolutely

boelle, 2019-07-21:
closing as the PR has been merged

closed 2019-07-21 by boelle