MarlinFirmware/Marlin #13702: [FR] Adding support for the Pyr0-Piezo Z-Probe (adding uart/i2c comms)

pyr0ball, 2019-04-14:
### Description

I've been working on a self-calibrating piezo sensor for a while, and I'm nearing completion, so I wanted to reach out and start working on code integration with Marlin. 

Information and sources can be found here: https://github.com/pyr0ball/pyr0piezo

### Requirements from Marlin

The sensor has been designed with UART and I2C input to allow the user to adjust the auto-calibration parameters on-the-fly. These sensors imitate a standard active-low endstop signal for it's output, so they can work out-of-the-box without these inputs, but there are a number of possible benefits for being able to adjust the sensitivity of the circuit:

- Different sized piezo disks produce different amplitude signals
- Decrease sensitivity during X/Y movements while probing, increase once probing Z
- Different mechanical structures for the piezo mount effect the signal amplitude

This integration could be implemented using GCode commands, and/or an LCD menu option

#### Input Syntax

To set the auto-tuning parameters using serial input, use the following:

* To change trigger active duration: TRG_D [integer for milliseconds]
* To change gain factor: GAIN_F [integer for gain state - see note*]
* To change sensor input pullup vRef high threshold: VADJH [float value]
* To change sensor input pullup vRef low threshold: VADJL [float value]
* To change comparator trigger high threshold: VCOMPH [float value]
* To change comparator trigger high threshold: VCOMPL [float value]

These commands should be wrapped in this format: 
`<CMD, INT, FLOAT>`

You must include the unused variable for each instance. 

Examples:
`<GAIN_F, 3, 0.00>`
`<VADJH, 0, 2.35>`

*Note for Gain Factor:

The gain STATE is representative of these values:
* 0 = 3x
* 1 = 3.5x
* 2 = 4.33x
* 3 = 6x
* 4 = 11x

InsanityAutomation, 2019-10-15:
Just dropping a note here to subscribe to the thread, as weve had some discussion on discord as well.

pyr0ball, 2021-01-24:
Just dropping this here as it's been a bit and I've added more documentation as well as an I2C interface chart: https://docs.pyroballpcbs.com/tutorials/config/pyr0piezo-i2c/pyr0piezo-i2c/