MarlinFirmware/Marlin #13704: [Feature Request] Auto set z offset

mattschinkel, 2019-04-15:
Every time you Change nozzle size, you need to set your z-offset if you have an auto level sensor such as bltouch.

I suggest adding a feature to auto set your z-offset by testing the height of the nozzle via pressing a button, then testing your probe and calculating the difference in height. This could be done before every print.

robbycandra, 2019-04-15:
I think this is good idea

InsanityAutomation, 2019-04-15:
This is doable if you dont use z probe as endstop. Lulzbot has configured the Taz 6 machines this way. Functionality has been supported for a long time.

robbycandra, 2019-04-15:
For user, the problem is testing the height of nozzle to the table. Need another probe to do that.

tcm0116, 2019-04-23:
If you have a button at at fixed offset relative to the bed, then you don't need a probe on the nozzle. Just have the nozzle home to the button and set your Z offset from there. If you change the nozzle, then it'll still home to the button, which has the same offset to the bed.

mattschinkel, 2019-04-23:
But them you don't have auto level

robbycandra, 2019-04-25:
@tcm0116 , but we have 2 unfix part here. Nozzle (because we can change nozzle) and bed (because we can adjust bed corner height).

Therefore, it is difficult to have a buttoh that have fixed offset relative to bed. Cause the bed is not fixed to the frame.

maukcc, 2019-08-09:
A good solution is the G31 Zxx.x code that repetier uses.
Is the Z_PROBE_OFFSET_FROM_EXTRUDER used for anything at this moment?

damvcoool, 2019-12-10:
In the SKR Mini E3 v1.2 you can configure the Z-Endstop and Probe on different Pins, wouldnt it be possible to simply calculate the difference between the Z-Endstop and when the probe gets triggered, to set the Nozzle to Probe Z-Offset?

qwewer0, 2021-01-01:
@mattschinkel `PROBE_OFFSET_WIZARD` should be helpful to get z-offset quickly.

sjasonsmith, 2021-01-01:
> `PROBE_OFFSET_WIZARD` should be helpful to get z-offset quickly.

I am actually going to close this, since I think the probe offset wizard fulfills this request.

closed 2021-01-01 by sjasonsmith