MarlinFirmware/Marlin #13705: [bugfix 2.0] Button not working for pause on stow/deploy with G29 P1 (same with #13114)

robbycandra, 2019-04-15:
Nothing happened when user push the button. 
Problem in probe.cpp -> probe_specific_action

```
        PGM_P const ds_str = deploy ? PSTR(MSG_MANUAL_DEPLOY) : PSTR(MSG_MANUAL_STOW);
        ui.return_to_status();       // To display the new status message
        ui.set_status_P(ds_str, 99);
        serialprintPGM(ds_str);
        SERIAL_EOL();

        KEEPALIVE_STATE(PAUSED_FOR_USER);
        wait_for_user = true;
        #if ENABLED(HOST_PROMPT_SUPPORT)
          host_prompt_do(PROMPT_USER_CONTINUE, PSTR("Stow Probe"), PSTR("Continue"));
        #endif
        while (wait_for_user) idle();
        ui.reset_status();
        KEEPALIVE_STATE(IN_HANDLER);
```

jackwilsdon, 2019-04-18:
I can confirm this is an issue for me too.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

robbycandra, 2019-09-24:
@boelle , never try again.

timeburn, 2019-11-18:
This is still an issue in the latest, as of today.  ABL is fine, UBL hangs at "Deploy Z-Probe"
Curiously, the deploy/stow works fine for Delta Calibration, just not UBL

sjasonsmith, 2019-11-18:
I can also confirm this issue exists, at least on delta. I didn’t realize there was already an issue reported, it was on my lift to report. @boelle, can you re-open this?

sjasonsmith, 2019-11-18:
I guess there are others open now too, so as long as one is open we should be good.

timeburn, 2019-11-24:
This was solved under issue 15927.  my bad for opening a dupe, but it's fixed now

boelle, 2019-12-02:
closing since its solved 

> This was solved under issue 15927.

closed 2019-12-02 by boelle