MarlinFirmware/Marlin #13706: Marlin 2.0 LPC1768 MKS_SBASE MAX6675 [FR]

pediegoivan, 2019-04-15:
I have been trying to get the MAX6675  working with a MKS SBASE running Marlin 2.0.x branch.
I saw that the MAX6675_SS_PIN was not defined in the PINS_MKS_SBASE file so tryied many pins with no luck.
It compiles ok, but when the board flashes the firmware.bin the leds stop flashing and it stays without serial connection, and If I plug the usb back there is no usb access to the on board sd card.
I assume that is a problem related with the on board spi that get the access to the SD card.
I was trying to get the MAX6675 working from the J7 plug as this pin arrangement seems convenient for this board.

I hope someone can point me on the right direction so I figure how to do it.
Thanks.

pediegoivan, 2019-04-15:
I have an update. I disabled everything I could find that could cause SPI incompatibilities, so there is not LCD SD or anything else.
Now I am able to connect but as soon as the board finishes booting up, It says MAXTEMP alarm was triggered and the board freezes and disconnects.

pediegoivan, 2019-04-15:
Ok. I finally made it work.
I could only make it work by dissabling any LCD or SD card support and adding the pin declaration for the MAX6675 for the J7 header.
I am sorry that I cannot make it further progress with this. I hope someone helps me to keep the other functionalities so we can add this lines to the pins_MKS_SBASE.h the proper way to add it to the Main code.

timh0r, 2019-04-16:
you should read the pins file for this board and the board manual, both clearly say / show that J7 is a bad idea because it's pins are shared with other connectors. Even J8 doesn't work all the time

pediegoivan, 2019-04-16:
I did! I am not using an external lcd. I am just using the motherboard. There are no suggested pins for running MAX6675 on this board. J7 pins are shared with the EXP1 connector that I am not using for anything else.

boelle, 2019-07-21:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

sjasonsmith, 2020-11-10:
@pediegoivan is this still something you are interested in? At this point I'm assuming this has been resolved. I'm going to remove the Feature Request tag and automatically close after a week if we don't hear anything back.

pediegoivan, 2020-11-10:
I swapped out the sbase board, so I suppose it is solved. I never was able.to mske it work. I just didn't use an lcd at all.

sjasonsmith, 2020-11-10:
Ok, thanks for letting us know. I’ll close this, as we don’t know if there is still an issue and you are no longer impacted by it.

closed 2020-11-10 by sjasonsmith