MarlinFirmware/Marlin #13707: [BUG] Marlin 2.0 Bugfix won't compile for BIQU SKR v1.1 with basic configuration

FelixHauser, 2019-04-15:
### Description

Marlin 2.0 Bugfix won't compile with a BIQU SKR v1.1 board

### Steps to Reproduce

1. Download latest Marlin 2.0 Bugfix (as of 15 April 2019)
2. Open up configuration.h and change the board to:  BOARD_BIQU_SKR_V1_1
3. Open up platformio.ini and change env_default to: LPC1768
4. Go ahead and compile

**Expected behavior:** [What you expect to happen]

I expected to compile without problems as I have not yet modified anything major

**Actual behavior:** [What actually happens]

It does not compile

#### Additional Information

So I am modifying my Mierder-3 printer and I changed the board to a BIQU SKR v1.1. I do have the same board for an Anycubic Kossel and it works just fine, without problems and I could compile its firmware just nicely.
I download a fresh copy of Marlin 2.0 Bugfix, since they added new features. The only two things I changed are, as above mentioned, the Board definition on configuration.h and the environment on platformio.ini. The firmware won't compile no matter what. It's throwing errors regarding a Neopixel thing. I checked on configuration.h and, although I see this somewhere, it's not actually defined anywhere. See screenshot of the error for more info.

Any help will be much appreciated!

[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/3078831/Configuration_adv.txt)
[image]
[platformio.txt](https://github.com/MarlinFirmware/Marlin/files/3078832/platformio.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/3078833/Configuration.txt)

doggyfan, 2019-04-15:
It's compiling for me OK (using a Marlin v2 snap shot from last week) using your config files.

Are you allowing platformio to download the various packages etc from the net when it compiles ? .. ie. not being blocked by an OS firewall or router firewall.

FelixHauser, 2019-04-15:
I don't think so, no... thanks for trying it...

RudyBenner, 2019-04-15:
Example file used, compiles fine here. I will try your files.

RudyBenner, 2019-04-15:
try it like this

[image]

instead of 

[image]

FelixHauser, 2019-04-15:
Well, I just tried with another computer and it seems to work...

p3p, 2019-04-15:
@FelixHauser Compilation is already as easy as possible, install PlatformIO, setup config and build.

Make sure your libraries and frameworks are up to date, this error means your building with framework-arduino-lpc176x @ 0.1.1, 0.1.2 fixes it.

FelixHauser, 2019-04-15:
> Make sure your libraries and frameworks are up to date, this error means your building with framework-arduino-lpc176x @ 0.1.1, 0.1.2 fixes it.

well, this is exactly what I meant... I don't understand any of what you just said... I know arduino and I know LPC1768, that's it

p3p, 2019-04-15:
Even with Arduino you need to install and keep things updated, this isn't part of Marlin,

Copied from #13329
> Depends how you use PlatformIO to build Marlin.
> 
> Terminal:
> platformio update or pio update
> 
> Atom:
> Click the PlatformIO menu then select Update platforms, packages and libraries
> 
> Visual Studio Code:
> Click the Platform IO symbol (Alien head) on the left menu bar, in Quick access Menu click on Updates -> Update All
>

FelixHauser, 2019-04-16:
that was helpful!
Thanks!

closed 2019-04-16 by FelixHauser