MarlinFirmware/Marlin #13708: Delta BLTouch auto bed leveling issue

IronDeltA, 2019-04-15:
### DESCRIPTION

When I try to use **bilinear bed leveling** on my FLSUN delta printer, I get the message "(L,R,F,B) out of bounds". When I try to define the bounds in repetier, using the command G29 L-29 R50 F-26 B50, it probes only a quarter of the bed. I've tested some changes I've found that worked for people with the same issue but, infortunately, it didn't work for me. I'm using **Marlin-AC-1.1.9-AC.**

Attached files: 
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/3080030/Configuration_adv.h.txt)
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/3080031/Configuration.h.txt)

shitcreek, 2019-04-15:
"(L,R,F,B) out of bounds" - this message is returned by G29 when the position is not reachable by the probe. Is your probe within bound?
You might want to look into G33: http://marlinfw.org/docs/gcode/G033.html
with S2 to save to EEPROM

IronDeltA, 2019-04-15:
Well, the G33 command works correctly, but it depends on the delta printable radius and not on the calibration radius, is this normal ? Also, for my configuration when I try to use the G29, the offsets of my probe are X-21 and Y-34, I defined delta printable radius at 80 and calibration radius at 65, and I also defined bounds:

Configuration.h extract:

  #define LEFT_PROBE_BED_POSITION -(DELTA_PRINTABLE_RADIUS - 21)
  #define RIGHT_PROBE_BED_POSITION (DELTA_PRINTABLE_RADIUS - 10)
  #define FRONT_PROBE_BED_POSITION -(DELTA_PRINTABLE_RADIUS - 34)
  #define BACK_PROBE_BED_POSITION (DELTA_PRINTABLE_RADIUS -10)
 
 #define MIN_PROBE_EDGE 10

mylife4aiurr, 2019-06-23:
This issue still exist on bugfix 2.0 3d9d72e

my config is here
https://github.com/MarlinFirmware/Marlin/issues/14347

boelle, 2019-09-24:
@IronDeltA & @mylife4aiurr is the issue still there?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle

epicurean70, 2019-10-23:
Can someone share their working configuration files for Malin 2.0 with bltouch enabled for a delta printer? I cannot get any combination to work for my anycubic linear plus

mylife4aiurr, 2019-10-25:
I sorta moved on to other projects but i can post my delta config from around august 2019:
Delta skr v1.3 Marlin 2.0x- 2971b48

MOTHERBOARD BOARD_BIGTREE_SKR_V1_3
#define DELTA_AUTO_CALIBRATION
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define BLTOUCH
#define X_PROBE_OFFSET_FROM_EXTRUDER -9   // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 35   // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER  0   // Z offset: -below +above  [the nozzle]

#define AUTO_BED_LEVELING_UBL
#define RESTORE_LEVELING_AFTER_G28

#define BLTOUCH_HS_MODE

[config.zip](https://github.com/MarlinFirmware/Marlin/files/3774179/config.zip)