MarlinFirmware/Marlin #13709: [BUG] BLTouch repeated probing into stopping

mrdomo, 2019-04-15:
### Description

Genuine ANTCLABS BLTouch Smart v3 probes into bed and when it hits the bed it retracts but redeploys and slowly gets pushed back in.

### Steps to Reproduce

1. Install BLTouch
2. Home

**Expected behavior:** Probe hits bed, probe retracts and records data as z home

**Actual behavior:** Probe hits bed, probe retracts, redeploys, slowly goes down into bed, retracts back in, BLTouch blinks red and blue with probe still deployed

#### Additional Information

I've triple checked the wiring, I've used a fresh install of marlin following a [video on how to install the BLTouch on the Ender 3](https://www.youtube.com/watch?v=sUlqrSq6LeY). This is a new BLTouch that just arrived today.

Here's a [video of the issue](https://youtu.be/YaFVA8hjfpU)
And my [configs](https://github.com/MarlinFirmware/Marlin/files/3080712/Configs.zip)

shitcreek, 2019-04-15:
I think you skipped this bit: https://youtu.be/sUlqrSq6LeY?t=356
#define Z_MIN_PROBE_ENDSTOP_INVERTING **false** // **set to true to invert the logic of the probe**.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-09-24:
@mrdomo & @Aurawin is the issue still present?

Aurawin, 2019-09-24:
There have been many improvements over the past few months.  The logic is getting very smart.  I've only seen this happen once recently during a g34 dual axis align.  

My left stepper mashed my bed up on that side.  It didn't stop.  This must have been some sort of calculation issue.  I would say that my right side was probably much higher than the algorithm could produce a valid left compensation and overshot.  Since then I've placed printed stoppers on each lead screw.  This makes sure the nozzle doesn't reach the actual build plate.  But every once in awhile alignment does mash into the lead stoppers. nI could manually skew my right side up to confirm...

That said, theoretically (and expectedly) if the bed is too high on 1 side, the nozzle would mash if the homing sensor was offset enough.  In that case, the setting for Z_PROBE_LOW_POINT should be reduced?

closed 2019-10-12 by boelle