MarlinFirmware/Marlin #13713: RADDS + EXPANSION V3 + REPRAP_DISCOUNT_SMART_CONTROLLER

pico01, 2019-04-15:
I have running this Combination with 4 Extruders and Heatbed. When Compiling I get this Error(s):
----------------------------------------------------------------------------------------------------
Arduino: 1.8.9 (Windows 10), Board: "Arduino Due (Programming Port)"
sketch\src\lcd\dogm\status_screen_DOGM.cpp: In function 'void _draw_heater_status(int8_t, bool)':
sketch\src\lcd\dogm\status_screen_DOGM.cpp:119:28: error: 'STATUS_BED_TEXT_X' was not declared in this scope

   const uint8_t tx = IFBED(STATUS_BED_TEXT_X, STATUS_HOTEND_TEXT_X(heater));
                            ^
sketch\src\lcd\dogm\status_screen_DOGM.cpp:108:34: note: in definition of macro 'IFBED'

     #define IFBED(A,B) (isBed ? (A) : (B))
                                  ^
exit status 1
--------------------------------------------------------------------------------------
with 3 Extruders or 4 Extruders and Heatbed=0 it compiles without Errors.

Think there ist too less Space on LCD for all the Symbols, but don't know how to fix this.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

shitcreek, 2019-08-16:
Take a look inside 'dogm_Statusscren.h' and you will see how there isn't anything set up for 4 hotends and a heated bed - the most is this:
`#if HAS_HEATED_BED && HOTENDS <= 3`
You can add your own condition and cram the graphics for 4 hotends and a heated bed. You'll need to reposition the corresponding temperature outputs.

boelle, 2019-09-24:
@pico01 is this still a problem?

pico01, 2019-09-26:
I'm not that good in programming, as I can schange all the Stuff that needed for the Change.
At the Moment, I reduced the Numbers of Hotends to 3 and use he 4th in the Kragen for a Sensor to Z-Leveling. I'm not that good in programming, as I can schange all the Stuff that needed for the Change.

boelle, 2019-10-21:
i will close this one as it not a bug as such

please open a [FR] instead

closed 2019-10-21 by boelle