MarlinFirmware/Marlin #13715: Bug regarding PID management ? Or thermistor table issue ?

cstavaru, 2019-04-15:
Hello everyone,

I just installed an E3D V6 (genuine) hot end (Semitec 104GT2 thermistor, 25W heater cartridge with 5.4 ohm resistance) on a Creality CR10S printer, set it up in configuration.h (as HEATER_0_TEMP 5) and  at first I noticed that the PID autotune fails at most temperatures (tried 200 and 248C) , in the sense that if I use the suggested values, the temperature oscillates by around 10 C. I then tried to do a manual PID tuning, and couldn't seem to do it (any values I chose would result in the same oscillatory behavior).

But after more trials I noticed that oscillations appear only on some temperature ranges (for example 242-249C), but they disappear on immediately before or after values (works on 240C, fails at 242C, works at 250C but fails at 248C), by using the same PID values (P=20, I=1, D=10 seems a good combo for my system). However, auto PID tuning fails at most temperatures, even at 250 or at 240C (the returned values result in temperature oscillations).

So I wonder if this is a bug in PID management, a bug in the thermistor table, or anything else entirely ?

BTW, I am using the '1.1.x bugfix' branch.

Below you will find an illustrative picture of the issue. 

Thanks !

[image]

ManuelMcLure, 2019-04-16:
Try bugfix-2.0.x - there have been a lot of changes to the temperature code and that may have fixed your problem.

cstavaru, 2019-04-17:
> Try bugfix-2.0.x - there have been a lot of changes to the temperature code and that may have fixed your problem.

Just tried version bugfix-2.0.x - absolutely the same behavior.

It doesn't really bother me much because I can just print at 250C insead of 248C, but to me it looks that there is a software issue that could hide other temperature issues.

cstavaru, 2019-04-21:
Ok, so I figured out the issue: 

First, the PID_FUNCTIONAL_RANGE (the maximum temperature difference from target that the PID is active at) of 10 (the default in configuration.h) is not enough for this setup, temperature fluctuates so badly during initial PID settling that it reaches the limits of this range and causes the PID algorithm to be reset, and it starts over, so it never settles or it settles by chance. So I have increased the PID_FUNCTIONAL_RANGE to 14.

However, automatic PID tuning still fails. The correct PID values I have found are P=7 (quite low), I=1, D=40. I suspect the automatic PID tuning starts at higher P values or something, so a low P is not tried, even if it is may be the only value that works. Higher values result in uncontrollable oscillation no matter what.

So maybe the authors should consider a wider range for the P values when doing PID tuning.

Now my system is rock solid and has variations of at most 1 degree, at any target temperature.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

InsanityAutomation, 2019-07-23:
Keep in mind a number of these boards had an incorrect capacitor value throwing off temp readings as reverence voltage changed. May be a contributing factor.

cstavaru, 2019-07-23:
> Keep in mind a number of these boards had an incorrect capacitor value throwing off temp readings as reverence voltage changed. May be a contributing factor.

Indeed, but I have changed mine in the first day with the correct one. And it was rock solid with the original hot end. It is also rock solid and stable with my manually-found values. I still believe the auto-PID algorithm does not cover enough range of values and that's why it is not able to find a working P, I, D combination.

InsanityAutomation, 2019-07-23:
I agree its not optimal and could be improved. Ive never looked at auto tuned pid values as more than a starting point.

Regardless, unless someone wants to code multiple routine formulas and a method of selecting them, nowhere else to go from here on this one.