MarlinFirmware/Marlin #13720: [BUG] SKR 1.1 (and 1.3), Fysetc MINI12864 with Bugfix 2.0.x

axl12ap, 2019-04-16:
Tried to compile the daily bugfix 2.0.x (Apr 15 version) of Marlin with SKR 1.1 (also 1.3) motherboard and the new Fysetc MINI12864 but errors on compilation.

The main issue is that isn't define the correct pins in the related .H of these motherboards.
In pins_RAMPS.H we have this code:

    #elif ENABLED(FYSETC_MINI_12864)   // Added in Marlin 1.1.9+

      // From https://wiki.fysetc.com/Mini12864_Panel/?fbclid=IwAR1FyjuNdVOOy9_xzky3qqo_WeM5h-4gpRnnWhQr_O1Ef3h0AFnFXmCehK8
      #define BEEPER_PIN        37
      #define LCD_RESET_PIN     23

      #define DOGLCD_A0         16
      #define DOGLCD_CS         17

      #define BTN_EN1           31
      #define BTN_EN2           33
      #define BTN_ENC           35

      #define SD_DETECT_PIN     49

      #ifndef RGB_LED_R_PIN
        #define RGB_LED_R_PIN   25
      #endif
      #ifndef RGB_LED_G_PIN
        #define RGB_LED_G_PIN   27
      #endif
      #ifndef RGB_LED_B_PIN
        #define RGB_LED_B_PIN   29
      #endif

and in pin_BIQU_SKRV1.1.H we have only this code:

#if ENABLED(ULTRA_LCD)
  #define BEEPER_PIN       P1_30
  #define BTN_EN1          P3_26
  #define BTN_EN2          P3_25
  #define BTN_ENC          P2_11

  #define SD_DETECT_PIN    P1_31
  #define LCD_SDSS         P1_23
  #define LCD_PINS_RS      P0_16
  #define LCD_PINS_ENABLE  P0_18
  #define LCD_PINS_D4      P0_15
#endif

Please correct the missing selection of Fysetc MINI12864 in all boards.
Thank for supporting us guys.

gloomyandy, 2019-04-16:
Are you sure that the SKR V1.1 board will work with this display? I don't think that there are any pins connected to the EXP headers for the the DOG connections and I doubt if there are enough pins available for the RGB connections. On the V1.3 board again I'm not sure that there are pins available for the RGB functions (at least not without using pins that may already be defined for use in other areas). 

If you have the hardware and you have this working for a board you should submit a PR for the updated pins file (or at least post the details of a working configuration here). There is a very good chance that no one else has the hardware to test this.

axl12ap, 2019-04-16:
> Are you sure that the SKR V1.1 board will work with this display? I don't think that there are any pins connected to the EXP headers for the the DOG connections and I doubt if there are enough pins [...]

I checked the 1.1 and I can confirm 3 pin missing for RGB led control, so I think pretty incompatible.

On the newer 1.3 both connectors (EXP1/EXP2) are compatible with Fysetc Mini12864 LCD module:

SKR 1.3 connectors:
EXP1.1 = 5V
EXP1.2 = GND
EXP1.3 = 1.23
EXP1.4 = 1.22
EXP1.5 = 1.21
EXP1.6 = 1.20
EXP1.7 = 1.19
EXP1.8 = 1.18
EXP1.9 = 0.28
EXP1.10 = 1.30 

EXP2.1 = NC
EXP2.2 = GND
EXP2.3 = RESET
EXP2.4 = 1.31
EXP2.5 = 0.18
EXP2.6 = 3.25
EXP2.7 = 0.16
EXP2.8 = 3.26
EXP2.9 = 0.15
EXP2.10 = 0.17

Fysetc MINI12864 connectors:

EXP1.1 = 5V
EXP1.2 = GND
EXP1.3 = BLUE
EXP1.4 = GREEN
EXP1.5 = RED
EXP1.6 = LCD RST
EXP1.7 = LCD A0
EXP1.8 = LCD CS
EXP1.9 = BTN_ENC
EXP1.10 = BEEP

EXP2.1 = KILL
EXP2.2 = GND
EXP2.3 = RST
EXP2.4 = CD
EXP2.5 = MOSI
EXP2.6 = BTN_ENC2
EXP2.7 = SS
EXP2.8 = BTN_ENC1
EXP2.9 = SCK
EXP2.10 = MISO

nebman, 2019-04-16:
Check #13395 for Fysetc Mini 12864 with SKR v1.3

closed 2019-05-06 by axl12ap