MarlinFirmware/Marlin #13722: immediate crash with correct fan pin assignement / STM32F103V

hobiseven, 2019-04-16:
What happens : Code running on alfawise STM32F103VE board, with LCD + touch. 
All is normal, until we assign the fan_pin to PA15, which is the correct pin assignement. This creates a complete CPU crash. 

-Assigning a soft PWM for fan management gets the SW to run, but then we get random errors on heating bed and extruder 1 ( there is only one extruder, which should be 0...?) requiring reboot. 

Attached the Config files, as well as HAL_Timers_STM32F1.h 
[Alfawise_STM32F103_files.zip](https://github.com/MarlinFirmware/Marlin/files/3085313/Alfawise_STM32F103_files.zip)

This is the timer definition in the timer_STM32F1.H file below. Are the 3 timer assignement correct? Specifically PULSE_TIMER_NUM STEP_TIMER_NUM ???

@jmz52 , you may confirm.... 

define HAL_TIMER_RATE         (F_CPU)  // frequency of timers peripherals

#define STEP_TIMER_CHAN 1 // Channel of the timer to use for compare and interrupts
#define TEMP_TIMER_CHAN 1 // Channel of the timer to use for compare and interrupts

#if defined(MCU_STM32F103CB) || defined(MCU_STM32F103C8)
  #define STEP_TIMER_NUM 4 // For C8/CB boards, use timer 4
#else
  #define STEP_TIMER_NUM 5 // for other boards, five is fine.
#endif
#define TEMP_TIMER_NUM 2  // index of timer to use for temperature
#define PULSE_TIMER_NUM STEP_TIMER_NUM

#define TEMP_TIMER_PRESCALE     1000 // prescaler for setting Temp timer, 72Khz
#define TEMP_TIMER_FREQUENCY    1000 // temperature interrupt frequency

jmz52, 2019-04-16:
I assume you are using `board = genericSTM32F103VE`
Pins functions for this board are defined in `.platformio\packages\framework-arduinoststm32-maple\STM32F1\variants\generic_stm32f103v\board.cpp` file.
PA15 definition is
`{&gpioa,   NULL, NULL, 15, 0, ADCx}, /* PA15 */ `
and is does not configured as PWM capable pin - timer and channel are set to NULL.

Timer2 channel 1 primary pin is PA0, PA15 is an 'Alternative Function' accessible via AF remap.
To use PWM on PA15 you need to redefine timer2-related pins in boards.cpp file.
There is `[env:black_stm32f407ve]` build environment  that uses custom board definition for STM32407VE board. Script used is this environment was designed for STM32 Core framework but should be reusable for libmaple framework. Please contact me if it not.

Also I never tested if maple-based libraries support 'Alternative Function' for PWM.
If they don't you will need to switch to HAL_STM32 and STM32 Core libraries.

jmz52, 2019-04-16:
And  you need to change
`#define TEMP_TIMER_NUM 2`
as it conflicts with PA15 PWM that uses the same timer #2.

hobiseven, 2019-04-16:
@JMZ52 Yep, will do. Cacaotor or Tanguy, working with me will test all that tonight! I have latest 5.3.0 + latest Marlin Pull running now, TS OK, adjustments needed on the screen, but we will test all extensively.

thinkyhead, 2019-04-16:
@jmz52 — Do you have some ideas how we can make sure timers and pins avoid conflicts like this?

tpruvot, 2019-04-16:
indeed, timer2 in the board.cpp seems reversed with timer5 (cf the Datasheet) this board.cpp seems the same for most STM32F1 variants... may be for the smaller chip variants ? Ours is a 100pins one

hobiseven, 2019-04-17:
@jmz52 We are working on fixing the issues with @tpruvot 

 One question I have about this "limitation" : If we declare 

#define FAN_PIN          PA15 // PA15 or 77   // FAN
#define FAN_SOFT_PWM            // Required to avoid issues with heating or STLink
#define FAN_MIN_PWM        35   // Fan will not start in 1-30 range
#define FAN_MAX_PWM        255

This should in principle avoid the conflict above with Timer2, right? 

With this above declarations, we have from one board to an other different behaviors, but this time on the heater outputs. On some board, they simply do not work at all, or one, not the other... This might be a separate issue, or again a PWM mix...

This family of boards has a few revisions, and we try to identify also why they behave differently sometimes. All boards are using the same CPU, and same pinout around the CPU. We tested 6 boards so far. 
Any hint? 

We could provide you a board if you could also check...

jmz52, 2019-04-18:
@thinkyhead
I would suggest to add optional timers definition to \<board_name\>_pins.h file to keep board-specific configuration is one file.
HAL_STM32 already supports custom timers configuration and it can be reused for maple-based STM32 HALs.
HAL_STM32\HAL_timers_STM32.h: 
```
#elif defined STM32F1xx

  #ifndef STEP_TIMER
    #define STEP_TIMER 4
  #endif

  #ifndef TEMP_TIMER
    #define TEMP_TIMER 2
  #endif
```

jmz52, 2019-04-18:
@hobiseven
Can you give me a link your board's pinout data and source code including platformio.ini?
I have STM32F103ZE devboards board I can use to test this PA15 issue.

hobiseven, 2019-04-18:
@JMZ52, just sent you an invite to my git ( there are also binaries, it is a bit a mess, sorry. ) . 
We so far use fan soft PWM to run the fan instead of the real timer. 
AND
all the strange behaviors, unpredicable stuff on heaters is now solved. This was caused by the heater, extruder and bed pins being push pull, and not open collector. We need to still change that also. 

Finally, The last cosmetic issue is to display the buttons pictures. 
Touch screen is in, and works, with soft SPI, as HW SPI cannot be used, as the board designer mixed MOSI and MISO on SPI2... 

We just tested again all on 6 boards. So far, so good. 

As soon as all that is fixed, I will make a clean git public.

hobiseven, 2019-04-18:
@JMZ52,
Just to clarify : 2 heater pins are open drain now. and OK. Fan pin is push pull, and needs to be open drain too.

tpruvot, 2019-04-18:
else, yep, the STM32F4 100 pins is almost pin compatible with our F103VET6, only faster and 2 pins are differents... but didnt check all the pin features neither.. Lerdge K Board is one but 144 pins, so a bit different about that (but the 100 pins variant in the same datasheet is really similar)

tpruvot, 2019-04-18:
the Lerdge X is a 407VET6... the 100 pins variant

xC0000005, 2019-04-20:
The Lerdge K is also a 407VGT6, at least the one I used to crack their encryption is.

tpruvot, 2019-04-20:
yep, 144pins, vs 100 pins on the 407VET6 (Lerdge X)

tpruvot, 2019-04-20:
https://aliexpress.com/item/LERDGE-3D-Printer-Controller-Board-for-Reprap-3d-printer-motherboard-with-ARM-32Bit-Mainboard-control-with/32819068617.html

hobiseven, 2019-04-20:
@jmz52 , hope you grabbed the code. I have to share back the git with the 3 collaborator here. Made a few cosmetic changed in the UI...but it is not easy to put color in the Marlin icons.. Easy for the buttons, but not for the actual Marlin page.

hobiseven, 2019-04-21:
@jmz52 as I do not know how to co tact you beside git, I write it here. A big thank you from the French alfawise community for your work on stm32f1/ robin which allows us to start openning alfawise to open source and MARLIN! See https://www.lesimprimantes3d.fr/forum/topic/18260-alfawise-u20x-u30-marlin-2x-firmware-alternatif/

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-09-24:
@hobiseven is the issue still present?

tpruvot, 2019-09-25:
well, imo this one can be closed, the issue seems more a design/port remap one for our board... to be able to use the PA15 as hardware pwm pin

boelle, 2019-09-25:
oki will close then :-D we can always reopen if needed

closed 2019-09-25 by boelle