MarlinFirmware/Marlin #13724: [bugfix-1.1x] Unable to detect SD card at boot

yunstackCN, 2019-04-16:
### Description
The sd card can not be recognized when the power is turned on, but when the sd card is removed and reinserted, it can be recognized
### Steps to Reproduce
1. Turn on the 3D printer power.
2. Insert the sd card
3. LCD shows "Card inserted" and files in sd card can be find; 
     pronterface show "echo:SD card ok".
4. Turn off power and turn on again, LCD shows "No SD card"
    pronterface shows :
         Error:volume.init failed
[ERROR] Error:volume.init failed

Error:volume.init failed
[ERROR] Error:volume.init failed

Error:volume.init failed
[ERROR] Error:volume.init failed

Error:volume.init failed
[ERROR] Error:volume.init failed

**Expected behavior:** 
   Turn off power and turn on agagin, sd card can be detected.
**Actual behavior:** [What actually happens]
   Turn off power and turn on agagin, LCD shows "No SD card"
#### Additional Information
I have tried the following scenarios and they are invalid：
    1、add the  following code to end of setup() in marlin_main.cpp.
"#if ENABLED(SDSUPPORT)
  if (!card.cardOK) card.initsd();
#endif" 
    2、modify SPI SPEED, has tried each speed  
     #define SPI_SPEED SPI_HALF_SPEED
//#define SPI_SPEED SPI_QUARTER_SPEED
//#define SPI_SPEED SPI_EIGHTH_SPEED

[config.zip](https://github.com/MarlinFirmware/Marlin/files/3085780/config.zip)
I hope I can get your help, thank you.

psavva, 2019-04-17:
Have you tried another SD Card?

yunstackCN, 2019-04-17:
Yes, tried, by the way,my motherboard is ramps 1.4, the lcd controller is RepRapDiscount Smart Controller, and there is a RepRapDiscount Full Graphic Smart Controller, which is the same problem, maybe my configuration problem, but I want to know how to fix it， thank you

yunstackCN, 2019-05-08:
Changed the 16g Kingston card, the problem is solved

closed 2019-05-08 by yunstackCN