MarlinFirmware/Marlin #13725: Vase mode weird behavior

kaijrock23, 2019-04-16:
Hi, when I print in vase mode this happen
https://youtu.be/8rhoIhAdOjY
It happen only in vase mode tried cura and simplify same behavior with both.
Marlin 2.0 with preconfigured ender 3 files.
I always printed fine until today, vase mode too.

shitcreek, 2019-04-17:
Have you checked your connections?

kaijrock23, 2019-04-17:
Yes i did, I tried 3 different sd cards and other gcodes in vase mode, same results, I tried with USB connection and it printed without problem, just stopped for 1 sec for two time randomly but I read it is a simplify behavior glitch.

kaijrock23, 2019-04-17:
So I think that the sd slot is the problem.

robbycandra, 2019-04-17:
Ha.... @kaijrock23 , i know you print woman body from thingiverse. I print it with marlin 2.0 and no problem.
I guess it is your SD card problem, check sdcard Reader speed setting and speed class of your SDCard.

kaijrock23, 2019-04-17:
I will contact creality and have a new motherboard, clearly it's the card reader that suddenly stopped working, any suggestion for a good SD card for the ender 3?

robbycandra, 2019-04-18:
Try to change the SD Card first.

kaijrock23, 2019-04-18:
Done, same results.

Tymek, 2019-05-19:
I have the same problem with SKR v1.1. Only spiralize mode prints weird - other gcodes work fine. (from Cura & Slic3r)

Tymek, 2019-05-20:
It worked. I reconnected everything today and sliced with stock settings where only printing mode was changed.

boelle, 2019-07-21:
will close this one as it seems to have been a hardware issue

closed 2019-07-21 by boelle