MarlinFirmware/Marlin #13726: Is Smart RAMPS Supported In Marlin 2 yet?

Eddiiie, 2019-04-16:
Hello,
I have a Smart RAMPS and a Arduino Due.  I downloaded Marlin-bugfix-2.0.x, set the motherboard type to "BOARD_RAMPS_SMART_EEB", made settings match in configuration.h and configuration_adv.h from Marlin 1 bugfix and get compile errors.

I then started fresh, only setting motherboard type to BOARD_RAMPS_SMART_EEB and get compile errors.

It seems as I look closer it is complaining about an invalid motherboard setting.

Here are the errors

In file included from sketch\src\inc/MarlinConfigPre.h:31:0,

                 from sketch\src\inc/MarlinConfig.h:28,

                 from sketch\src\Marlin.h:24,

                 from sketch\src\Marlin.cpp:31:

sketch\src\inc/../core/boards.h:278:72: error: operator '==' has no right operand

 #define MB(board) (defined(BOARD_##board) && MOTHERBOARD==BOARD_##board)

                                                                        ^

sketch\src\inc/../pins/pins.h:40:96: note: in expansion of macro 'MB'

 #elif MB(RAMPS_13_EEB) || MB(RAMPS_14_EEB) || MB(RAMPS_PLUS_EEB) || MB(RAMPS_14_RE_ARM_EEB) || MB(RAMPS_SMART_EEB) || MB(RAMPS_DUO_EEB) || MB(RAMPS4DUE_EEB)

                                                                                                ^

sketch\src\inc/../core/boards.h:278:72: error: operator '==' has no right operand

 #define MB(board) (defined(BOARD_##board) && MOTHERBOARD==BOARD_##board)

                                                                        ^

sketch\src\inc/../pins/pins.h:362:7: note: in expansion of macro 'MB'

 #elif MB(RAMPS_SMART_EEB)

       ^

In file included from sketch\src\inc/MarlinConfig.h:32:0,

                 from sketch\src\Marlin.h:24,

                 from sketch\src\Marlin.cpp:31:

sketch\src\inc/../pins/pins.h:469:4: error: #error "Unknown MOTHERBOARD value set in Configuration.h"

   #error "Unknown MOTHERBOARD value set in Configuration.h"

    ^

In file included from sketch\src\inc/MarlinConfig.h:41:0,

                 from sketch\src\Marlin.h:24,

                 from sketch\src\Marlin.cpp:31:

sketch\src\inc/SanityCheck.h:56:4: error: #error "MOTHERBOARD is required. Please update your configuration."

   #error "MOTHERBOARD is required. Please update your configuration."

    ^

sketch\src\inc/SanityCheck.h:1003:8: error: #error "SERVO0_PIN must be defined for your servo or BLTOUCH probe."

       #error "SERVO0_PIN must be defined for your servo or BLTOUCH probe."

        ^
[... 192 lines cut ...]

   static const pin_t sensitive_pins[] PROGMEM = SENSITIVE_PINS;

                                                 ^

sketch\src\pins/sensitive_pins.h:125:40: error: 'Z_ENABLE_PIN' was not declared in this scope

 #define _Z_PINS Z_STEP_PIN, Z_DIR_PIN, Z_ENABLE_PIN, _Z_MIN _Z_MAX _Z_MS1 _Z_MS2 _Z_MS3 _Z_CS

                                        ^

sketch\src\pins/sensitive_pins.h:523:19: note: in expansion of macro '_Z_PINS'

   _X_PINS _Y_PINS _Z_PINS _X2_PINS _Y2_PINS _Z2_PINS _Z3_PINS _Z_PROBE \

                   ^

sketch\src\Marlin.cpp:262:49: note: in expansion of macro 'SENSITIVE_PINS'

   static const pin_t sensitive_pins[] PROGMEM = SENSITIVE_PINS;

                                                 ^

exit status 1
Error compiling for board Arduino Due (Programming Port).

psavva, 2019-04-17:
Please attach your config files

Eddiiie, 2019-04-17:
I looked again and saw that I was missing MOTHERBOARD on the define statement.  

I had - 
#define BOARD_RAMPS_SMART_EEB
NOT
#define MOTHERBOARD BOARD_RAMPS_SMART_EEB

Sorry for the false alarm.

closed 2019-04-17 by Eddiiie