MarlinFirmware/Marlin #13728: [FR] (feature request title)

co8fs, 2019-04-16:
### Description

### Steps to Reproduce

1. [First Action]
2. [Second Action]
3. [and so on...]

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Hi Guys

I am fairly new to 3D Printing and Programming ( I am learning).
I am trying to connect an MKS MINI12864 Display to a Rumba Board running Marlin or Repetier.
But so far is not working, I got it working with the ReprapDiscount Full Graphic Smart Controller.
Please anybody can "Help"
Thanks.

Sebastian
K4BAC

marcio-ao, 2019-04-17:
@co8fs: You might get more eyeballs on this is you change the title of the issue :)

co8fs, 2019-04-17:
Sorry marcio
that won't happen again. Sorry again.I thought this is an open platform to get help fro the community.Thanjks.

co8fs, 2019-04-17:
MarcioI already opened another topic in github.
    On Wednesday, April 17, 2019, 5:52:23 PM EDT, Sebastian Fernandez <co8fs@yahoo.com> wrote:  
 
  Sorry marcio
that won't happen again. Sorry again.I thought this is an open platform to get help fro the community.Thanjks.

marcio-ao, 2019-04-18:
@co8fs: Thank you for changing the title, it is much better. It will help the right people look at the issue. I don't have a MKS MINI12864 so unfortunately I cannot offer assistance, but perhaps someone else can.

boelle, 2019-07-21:
so this is a feature  request? or is it a question?

boelle, 2019-09-24:
@co8fs tried discord?

-Marlin Discord server. Join link: https://discord.gg/n5NJ59y

boelle, 2019-10-24:
better ask here: 

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

this list is for bugs and fr's only

closed 2019-10-24 by boelle