MarlinFirmware/Marlin #13732: MINIPANEL showing blank screen (Creality Ender 2) - Marlin 2.0

drofnas, 2019-04-17:
After merging the latest Marlin 2.0 bugfix codebase into my fork, I noticed that my Ender 2's screen was blank (lights on, no info). But the machine itself was still working via the connected Raspberry Pi running OctoPrint.

Since a previous version was working on my Ender 2, I re-installed that just to confirm my screen didn't die on me. The screen worked again.

So I went back in time and starting doing a "git checkout" of older commits to try and narrow down where the issue occurred. Which led me to the following commit, where the MINIPANEL and the FYSETC_MINI_12864_PANEL where lumped together for various config values. 

https://github.com/MarlinFirmware/Marlin/commit/5018fdacbeb0807fc255f83049044b89323eea55

If I get one commit older, my screen works perfectly again.

For all tests I was using the example "Creality / Ender 2" config files that were part of the specific commit ID's I was testing.

Idorobots, 2019-04-18:
Perhaps this is the same issue as https://github.com/MarlinFirmware/Marlin/issues/13550 ?

drofnas, 2019-04-18:
This is running on a Stock Creality Ender 2, using the stock Melzi Clone with a Atmega1284P chip. Plus the bug you reference is older than the commit I reference above; which is the point where my screen stopped working.

jmdearras, 2019-04-18:
> This is running on a Stock Creality Ender 2, using the stock Melzi Clone with a Atmega1284P chip. Plus the bug you reference is older than the commit I reference above; which is the point where my screen stopped working.

That bug was initially intermittent, but became solid a week or so ago.  With a blank display, when you hit reset, did the screen show text for a second?

drofnas, 2019-04-18:
Yea, I think I did see "something" for a second, then went blank.

jmdearras, 2019-04-21:
> Yea, I think I did see "something" for a second, then went blank.

Same issue.  See the other thread for a hopefully temporary patch.

zillarob, 2019-04-22:
I think this might be something else.
I rolled one up a few weeks ago that had symptoms that match what that temp fix is for, but the most recent one I did just shows a blank screen on the mks mini. Same frozen image with reset.

Just grabbed the few week old project, set it up for the new one, and works now. (I'm sure I will run into the shifted/mirrored screen occasionally though).
Both machines are SKR v1.3 with MKS Mini.

shitcreek, 2019-04-30:
See if this works: #13863

shitcreek, 2019-05-01:
Try this: #13864

dlitz, 2019-06-20:
I'm seeing the same thing on my CR-20 (which is a RAMPS-like board).  I posted some more info on the issue that @shitcreek referenced.

dlitz, 2019-06-20:
Changing the definition of `HAS_LCD_CONTRAST` caused it to evaluate true for `MINIPANEL`, but apparently the code that sets contrast values on MINIPANEL doesn't work correctly, at least not on the Ender 2 (or my CR-20---I ran into the same issue there).  See #14174 for a possible workaround.

boelle, 2019-07-21:
@drofnas figured this one out?

Diekion, 2019-08-07:
I find myself in the same situation. The screen turns on but does not display any information. I cannot provide prior information since the SKR 1.3 is the first pcb and I try to replace it from the original.

In case it helps.
What I have noticed, disconnecting wire by wire on the original board, is that the distribution of pins is like CR_10 except the equivalent of pin 1.30. If I disconnect it on the original board, the screen goes blank as it happens in the SKR and in this SKR pin 1.30 is configured for the beeper that does not have on this lcd board. I think it may be related to the configuration of that pin.

thinkyhead, 2019-08-14:
Let's move discussion over to #14174. And there should be a patch shortly.

closed 2019-08-14 by thinkyhead