MarlinFirmware/Marlin #13733: G29 printer skips probing certain spots and then prints in air

yashyboi, 2019-04-17:
every time i use auto bed leveling, the z axis probes the bed, starting normally, but when the probe moves to the second probing point, it stops going down to check the bed and just skips to the next probing spot after pausing. this means that once the print begins, some spots are perfectly printing, while others are printing in midair. i have already replaced my arduino and ramps, tried different versions of marlin, im not sure what the problem is. im pretty sure it has to do with marlin though.
it doesnt matter if i use UBL, bi linear or any of the other versions, they all do this.

reloxx13, 2019-04-17:
config?
z-offset?
terminal output while probing?
video of probing would be good.

robbycandra, 2019-04-17:
@yashyboi , the problem is in your z offset probe. Not marlin problem.

yashyboi, 2019-04-17:
@robbycandra if the problem was with the probe, why does it home the z axis perfectly? It reads open when it's not triggered and reads triggered when it should. The printer doesn't even bother bringing it close to the bed on the second probing spot, it just moved over to it, and then goes to the next spot, where the probing works properly again. I'll try to get you guys a video of it later today so you'll get what I'm talking about. I hope it's just some spooky probe problem though, I'd much rather just swap the probe. It also does kinda make sense that it'd be a probe problem as the problem arose after a few prints, not directly after a marlin upload. anyways thanks for the help, I'll try swapping out the probe and I'll get you guys a video today

RudyBenner, 2019-04-17:
Where are your config files?

yashyboi, 2019-04-17:
@rudybenner sorry this is my first post on here, how can I upload my config files?

RudyBenner, 2019-04-17:
Make a zip file of all the files you wish to submit, then copy/paste/drop whatever.

yashyboi, 2019-04-18:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3092253/Marlin.zip)

RudyBenner, 2019-04-18:
which version of marlin?

RudyBenner, 2019-04-18:
Line 555 - do you have a single DRV8825 and the others are A4988?
Not a BLTouch?
Line 886 887 - typically negative values.

yashyboi, 2019-04-18:
@rudybenner I have one drv8825 yeah
I'm not using a bltouch, I'm using an induction sensor
I'll check over those lines when I get home

RudyBenner, 2019-04-18:
If the probe has a status LED, watch it change state when it probes.

RudyBenner, 2019-04-18:
For example:

/**
 * Z probes require clearance when deploying, stowing, and moving between
 * probe points to avoid hitting the bed and other hardware.
 * Servo-mounted probes require extra space for the arm to rotate.
 *  _Inductive probes need space to keep from triggering early._
 *
 * Use these settings to specify the distance (mm) to raise the probe (or
 * lower the bed). The values set here apply over and above any (negative)
 * probe Z Offset set with Z_PROBE_OFFSET_FROM_EXTRUDER, M851, or the LCD.
 * Only integer values >= 1 are valid here.
 *
 * Example: `M851 Z-5` with a CLEARANCE of 4  =>  9mm from bed to nozzle.
 *     But: `M851 Z+1` with a CLEARANCE of 2  =>  2mm from bed to nozzle.
 */
#define Z_CLEARANCE_DEPLOY_PROBE   10 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  5 // Z Clearance between probe points
#define Z_CLEARANCE_MULTI_PROBE     5 // Z Clearance between multiple probes
//#define Z_AFTER_PROBING           5 // Z position after probing is done

#define Z_PROBE_LOW_POINT          -2 // Farthest distance below the trigger-point to go before stopping

// For M851 give a range for adjusting the Z probe offset
#define Z_PROBE_OFFSET_RANGE_MIN -8
#define Z_PROBE_OFFSET_RANGE_MAX 8

RudyBenner, 2019-04-18:
Make a graphic drawing to help visualize what you are aiming for.
Write stuff down. Make notes. Make backups.

yashyboi, 2019-04-18:
Sorry bout the wait guys, just waiting for a new stepper driver to come in because I accidentally killed one! (Stepped on it) as soon as it comes in I'll install it and upload a video of what is going on. Thank you all so much for the help

yashyboi, 2019-04-22:
(UPDATE) printer bed leveling now works correctly, the issue was the bed leveling probe! if you have a similar problem, get a new z probe!
thanks to everyone who helped, you guys are simply incredible.

robbycandra, 2019-04-22:
@yashyboi , you can close this issue

closed 2019-04-22 by Roxy-3D