MarlinFirmware/Marlin #13734: [BUG] TCM5160 is compiling without errors without CS pins defined

Svatevit, 2019-04-17:
### Description

I just installed TMC5160 on E0 and tried to compile firmware. It did without errors, but it's not working. When I changed E0 to TMC2130 in code, it says I don't have CS E0 pin defined.

### Steps to Reproduce

Remove/comment E0_CS_PIN from pin definitions, set TMC5160 as driver for E0.

**Expected behavior:** [E0_CS_PIN not defined error]

**Actual behavior:** [Compilation finishes succesfuly, printer dead]

teemuatlut, 2019-04-17:
Yea looks like the sanitychecks are only for TMC2130.
If you don't define it, Marlin will assign `-1` for the pin and there won't be a compile error from a missing define.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

InsanityAutomation, 2019-07-23:
This may be valid as some installations bring the pin directly to ground instead of software controlling it to make it work from my understanding. @teemuatlut In your opinion should a sanity check be added?

boelle, 2019-10-12:
i would say a sanity check should be added

boelle, 2019-10-24:
i add the confirmed label as it seems pretty clear that we need the sanity check

boelle, 2019-11-09:
@thinkyhead @shitcreek @teemuatlut would a sanity check not "solve" this one?

boelle, 2019-12-07:
@Svatevit since 2.0 was just released a few days ago has this changed this issue at all?

boelle, 2020-01-10:
pr has been merged so should be fixed

closed 2020-01-10 by boelle