MarlinFirmware/Marlin #13738: Compile error?

RudyBenner, 2019-04-17:
[image]

Changing the = to a + makes the problem go away.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3091029/Marlin.zip)

marcio-ao, 2019-04-17:
@RudyBenner: I've been seeing it too, but it's a warning, not a compiler error:

```
src/lcd/dogm/ultralcd_DOGM.cpp: In static member function ‘static void MarlinUI::show_bootscreen()’:
src/lcd/dogm/ultralcd_DOGM.cpp:185:31: warning: overflow in implicit constant conversion [-Woverflow]
       txt_offx_1 = txt_offx_2 = inter + (START_BMPWIDTH) + inter; // Text right of the bitmap
```

Changing = to + is probably not a correct fix :)

RudyBenner, 2019-04-17:
I would not be at all surprised if that is not a correct fix. Thnaks.

marcio-ao, 2019-04-17:
@RudyBenner : As far as I can tell, correct fix is to change this:

```
constexpr uint8_t inter = (width - text_max_width - (START_BMPWIDTH)) / 3; // Evenly distribute horizontal space
```

To this:

```
constexpr int8_t inter = (width - text_max_width - (START_BMPWIDTH)) / 3; // Evenly distribute horizontal space
```

RudyBenner, 2019-04-17:
i see more than one instance of uint8_t

RudyBenner, 2019-04-17:
I see 7 instances of uint8_t

closed 2019-04-18 by RudyBenner