MarlinFirmware/Marlin #13745: G29 dont work: L,R,F,B out of Bounds

arnold-b, 2019-04-17:
Hello, 
i m using current BugFix 1.1.x. 

If i send G29 i get just the Message: ? (L,R,F,B) out of bounds. 

I don´t find the reason. 

Thanks

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/3091730/Configuration.txt)

robbycandra, 2019-04-18:
Try Bugfix 2.0

arnold-b, 2019-04-18:
At least on this Version i don´t get the bounds error. But on starting G29 marlin Moves the Z Axis with some Feed and the stepper lose steps. After one second it use the defined Feedrate. But the position is already wrong and probing fails

reloxx13, 2019-04-18:
u set a max feedrate on z to 2000 and wonder why its skipping?...

if its configured and connected correctly, probing works fine with 1.1.x and 2.0.x

arnold-b, 2019-04-18:
If found the problem. The stepper was skipping because of XY_PROBE_SPEED in configuration.h 
I didnt know it exist and it was on 8000.  
I don´t know why suddenly the developers using the word "speed" instead of "feedrate". Just makes it more complicated.

Probe_pin didnt work, because i used pullup on this pin

closed 2019-04-18 by arnold-b