MarlinFirmware/Marlin #13747: [FR] set Heat control output pins as open drain

hobiseven, 2019-04-18:
Good Morning marlin team,

As we are debugging Marlin Alfawise port, I came to an issue about the heater control pins. 
We do need for those Alfawise boards open drain outputs, and as far as I can tell, no macro is available for that : 

#define SET_INPUT(IO) _SET_MODE(IO, INPUT) /*!< Input Floating Mode */
#define SET_INPUT_PULLUP(IO) _SET_MODE(IO, INPUT_PULLUP) /*!< Input with Pull-up activation */
#define SET_INPUT_PULLDOWN(IO) _SET_MODE(IO, INPUT_PULLDOWN) /*!< Input with Pull-down activation */
#define SET_OUTPUT(IO) OUT_WRITE(IO, LOW)
#define SET_PWM(IO) pinMode(IO, PWM)
 
Am I correct to state that Marlin assigns all pins PWM and outputs as regular outputs, driven low or high, but not open drain?

hobiseven, 2019-04-18:
All,
We checked and added an open collector output for the Alfawide printers, and the result is SUPERB. 
So, this is needed. 
Right now the fix is done in the fastIO_STM32F1.H

sjasonsmith, 2020-11-10:
@hobiseven are you still paying attention to this? Is that a change you could contribute back in a pull request?

slowbro, 2021-07-03:
I think this is resolved by #21159 which added options for BOARD_OPENDRAIN_MOSFETS and BOARD_PREINIT, for example..

https://github.com/MarlinFirmware/Marlin/blob/cd01421ac32041c7f775ec37dd8d00b29a5d335b/Marlin/src/pins/stm32f1/pins_LONGER3D_LK.h#L98-L106