MarlinFirmware/Marlin #13748: SKR 1.3 TMC2208 bad comms

doggyfan, 2019-04-18:
At startup Marlin v2 sends a byte (should be 4 ?) to the TMC2208 and receives an 8 byte zero response for an unknown reason.

Trying to attach the image of the UART comms from the scope to this message but it won't let me.

doggyfan, 2019-04-18:
The boards seem to be talking to the TMC2208 chips, but the SKR 1.3 board displays "TMC CONNECTION ERROR" on the LCD, and the TMC2208 is not configured properly. The MKS Gen board works fine with the same TMC2208's.

Any ideas ?

8-bit MKS Gen 1.4 ..`

[image]

32-bit SKE 1.3 ..

[image]

gloomyandy, 2019-04-18:
Many, many folks have TMC2208 working with Marlin on that board (it has been very popular) so I'm pretty sure that whatever problem you have is either down to your setup or hardware. Things to check...
1. Are you using the latest Marlin from the Marlin github (not the one from bigtreetech)?
2. Is you platformio up to date (you need to be using the latest LPC1768 framework).
3. Are you sure you have the jumpers on the SKR board configured correctly. Remove all jumpers from under the board, make sure that you have the UART jumpers in place (just in front of the driver boards). Post a picture if possible.
4. Make sure that you are powering your board using the power jumper set to "int" not to "usb" and that you have 12/24V supplied.
5. Are you sure that you have your driver boards setup correctly? There are many slightly different boards and each one needs a different configuration. In particular the SKR board uses the pin next to the PDN pin for UART comms, many other setups use the PDN pin. On some boards this is marked NC on others UART. Some driver boards require that you add a solder bridge between the PDN pin and the NC/UART pin. Just because your drivers work on the MKS board does not mean that are configured correctly for the SKR. Again post information as to which drivers you have and pictures of how you have them setup.

I've tried to look at your configuration files but I can't unzip them (how did you create the zip file?). Either way I would suggest that rather than using a heavily modified config you start with the basic Marlin files and make minimum changes to just get your drivers working, then add the complex stuff later.

doggyfan, 2019-04-18:
Yes to all 5 questions. That's what's stumped me, I didn't want to bother anyone on here but am at a loss at the moment.

I'll use clean config files to start with but am sure it won't make any difference, I'm many years development engineer in hardware and software and firmware etc, so not a boob that's for sure.

I'll get back to you guys shortly. I may have to try some debugging in the software serial driver code etc to get a better idea of what's going wrong, because it does look like the TMC2208 is replying, though not yet sure, I'll check the scope without the TMC2208 plugged in, see what the UART data line is like then.

[image]
[image]

doggyfan, 2019-04-18:
My bad :(

The SKR 1.3 board UART is fed to the N.C pin on the TMC2208 driver boards (I missed that difference).

My MKS Gen 1.4 board has the UART pin on the PDN pin on the TMC2208 driver boards.

Soldered the PDN to the N.C pin on each driver board and now the SKR 1.3 is working.

This thread can be deleted if you want. So sorry I bother bothered with this gloomyandy.

closed 2019-04-18 by doggyfan

gloomyandy, 2019-04-18:
No problem! Glad you found the cause. Yes it is a little unfortunate that they choose to use the "UART" pin rather then PDN. It has caused a lot of people issues!

doggyfan, 2019-04-19:
;)