MarlinFirmware/Marlin #13749: how to  activate POWER LOSS RECOVERY  in  bugfix-2.0.x

ahmedgk, 2019-04-18:
Hi
i use marlin bugfix-2.0.x.
I enabled the POWER_LOSS_RECOVERY

 #define POWER_LOSS_RECOVERY
  #if ENABLED(POWER_LOSS_RECOVERY)
    #define POWER_LOSS_PIN   44     // Pin to detect power loss
    #define POWER_LOSS_STATE HIGH   // State of pin indicating power loss
  #endif

I cut the electricity, I give it again, the printer home x and y and  then begin to reheat
the  Z rise and returned to correct X &Y but Z didnt get down again and start to print .

i try to uncomment this

#define NO_MOTION_BEFORE_HOMING
#define Z_HOMING_HEIGHT 0

it works but for one time when i try to cut electricity for the second time  it go to correct X Y Z and print one layer then Z rise

maxlinux2000, 2019-04-21:
Yes, we are trying to convince programmers that power_loss_recovery has to work always, and not just once.
Please look at these two bugs that I have opened, and comment your too, thanks

https://github.com/MarlinFirmware/Marlin/issues/13541
https://github.com/MarlinFirmware/Marlin/issues/13677

InsanityAutomation, 2019-04-21:
I think the convincing is more where it lands in the priority list than should be done... Right now it's just not at the top of the list. Too many of us just have too many projects open right now for anybody to be jumping on another!

boelle, 2019-07-21:
is this a bug or question?

boelle, 2019-10-24:
@ahmedgk you better ask here:

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

this list is for bugs and fr's only

closed 2019-10-24 by boelle

apaltd, 2020-04-12:
Hi, upgraded to Marlin bugfix2, un-commented  power_loss_recovery. not using any pin for power outage and after each layer marlin writes in PLR file on Sd card but after power outage and re plug, no resume print appears? any help will be appreciated.

thinkyhead, 2020-04-12:
@apaltd — Download the latest which has been patched.