MarlinFirmware/Marlin #13753: [bugfix-2.0.x] Drivers Detected but Steppers not spining

Andreiva13, 2019-04-18:
Hello!
So after (I think) I broke some TMC2130 I ordered and received some SilentStepSticks , now the drivers are detected after using M122 but if I try to use the steppers they don't spin , any ideas?
[config.zip](https://github.com/MarlinFirmware/Marlin/files/3095676/config.zip)
Here is the M122 response if it helps:
`Recv: 		X	Y	Z	E
Recv: Enabled		false	false	false	false
Recv: Set current	800	800	800	800
Recv: RMS current	795	795	795	795
Recv: MAX current	1121	1121	1121	1121
Recv: Run current	25/31	25/31	25/31	25/31
Recv: Hold current	12/31	12/31	12/31	12/31
Recv: CS actual		12/31	12/31	12/31	12/31
Recv: PWM scale	0	0	0	0
Recv: vsense		1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true
Recv: msteps		16	16	16	16
Recv: tstep		max	max	max	max
Recv: pwm
Recv: threshold		0	0	0	0
Recv: [mm/s]		-	-	-	-
Recv: OT prewarn	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false
Recv: off time		0	0	0	0
Recv: blank time	24	24	24	24
Recv: hysteresis
Recv: -end		-1	-1	-1	-1
Recv: -start		1	1	1	1
Recv: Stallguard thrs	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	E
Recv: stallguard
Recv: sg_result		0	0	0	0
Recv: fsactive
Recv: stst		X	X	X	X
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:
Recv: 		X	0x80:0C:00:00
Recv: 		Y	0x80:0C:00:00
Recv: 		Z	0x80:0C:00:00
Recv: 		E	0x80:0C:00:00
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing E connection... OK`

doggyfan, 2019-04-18:
What do you get from an M119 (Endstop states) ?

Andreiva13, 2019-04-18:
> What do you get from an M119 (Endstop states) ?

I get that all are open (as they should) , though I did not make the connections for sensorless  homing yet.

psavva, 2019-04-18:
Recv: X	0x80:0C:00:00
Recv: Y	0x80:0C:00:00
Recv: Z	0x80:0C:00:00
Recv: E	0x80:0C:00:00

Your wiring must be wrong.  These are supposed to give you different addresses, not the same...

teemuatlut, 2019-04-18:
@Andreiva13 Disable Software driver enable. It's not a fancy new feature to unlock dancing unicorns, but rather a necessity for some drivers that do not have a hardware EN line. An example is the Panucatt BSD2660.

A more technical answer to why the steppers are not moving is because the off time (toff) value is `0` which means the driver is disabled. This can also be seen from the `Enabled` debugging line that's there as the very first item.

> These are supposed to give you different addresses, not the same

They can be the same.

psavva, 2019-04-19:
Thank you for correcting me.  On my setup these  are totally different.  Isn't the definition of the address to identify each driver uniquely?

teemuatlut, 2019-04-19:
It's a 32bit read response with specific bits or bit patterns that tell of the status of the driver.

Andreiva13, 2019-04-19:
> @Andreiva13 Disable Software driver enable. It's not a fancy new feature to unlock dancing unicorns, but rather a necessity for some drivers that do not have a hardware EN line. An example is the [...]

This worked , thank you so much!

closed 2019-04-19 by Andreiva13