MarlinFirmware/Marlin #13758: [BUG] Unified Bed Levelling incorrect mesh height on edges (too high)

CorvetteCole, 2019-04-18:
### Description

Basically, I've probed the bed (all 225 points, 15 each axis) with a BLTouch. It does 3 slow probes and averages the data. I've got a 200x200mm print area but on a 220x220mm bed so let UBL probe the "edges" since they are actually solidly on the bed. The problem is, near the far right of the bed (and to a lesser extent the front and back) the mesh will say values like "+2.103", "+2.308" etc. I've ran the M48 repeatability test on one of the points that are far too high and I get values around .903 to 1.004. My probe's standard deviation is between 0.05 and 0.10. So my question is, how is UBL getting such different results from probing the same exact point with the same exact method? I have no idea what could be causing this but it is making printing anywhere but the middle of the bed impossible. My bed is fairly level without levelling, as in in this current state prints do better with no UBL at all vs UBL (still not amazing, warped bed).

### Steps to Reproduce

1. Run G28 (mine homes z in the middle of the bed with -0.3 offset
2. Run G29 P1 T
3. Position probe at a point on the far right of the bed
4. Run M48 
5. Compare position of M48 with the point found in G29 T

**Expected behavior:** They should be roughly the same, within standard deviations

**Actual behavior:** UBL is more than 1 mm off consistently

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3096405/Configuration.zip)

Roxy-3D, 2019-04-18:
I have several questions:

- How much memory is free when you build the firmware?   15 x 15 might be eating up too much memory.
- Where are homing the Z-Axis ?   It is best (especially for this debugging process) to home in the center of the bed.
- When you home the Z-Axis in the center of the bed, what number do you see on the LCD Panel for Z when slowly and carefully lower the nozzle so it just touches the bed?

CorvetteCole, 2019-04-19:
1. Arduino reports this: "Sketch uses 144806 bytes (57%) of program storage space. Maximum is 253952 bytes.
Global variables use 5509 bytes (67%) of dynamic memory, leaving 2683 bytes for local variables. Maximum is 8192 bytes.
"

2. I am homing it in the center of the bed already

3. I don't have an LCD, instead I am using repetier host which provides the same info and functionality. I have calibrated the probe z offset such that it shows 0 as Z when at the bed

4. I can, let me redo the mesh and ill get back to you

gloomyandy, 2019-04-19:
What happens if you do a G28 between probing the and and running the M48? Also can you trying doing the following, run G28,then move the probe to the same position you are using to run the M48,now run the G29 P1. This should result in a different probing order, how does the G29 T output compare with the results you got from the "normal" G29 P1?

CorvetteCole, 2019-04-20:
Hi, I found the problem, and it isn't with Marlin, but it isn't quite what you'd expect. The problem doesn't even lie within the BLTouch probe itself... Instead, with the 5v wires that power it. Turns out, the gauge wire I was using carries 5v fine but during the 100mA spike the sensor has when probing it gives really inaccurate values because of voltage drop in the wire. After connecting a buck converter to the 12v accessory lines already on the extruder, the standard deviation of the probe dropped to 0.006 and UBL worked as expected when creating a mesh.

My question now is why a mesh created with a probe standard deviation of 0.10 had such a large (1mm+) difference over the actual height probed by M48 at the time.

It may simply be that the probe was really much worse than the M48 standard deviation would suggest, or there may be a software issue somewhere as well. May be worth looking in to, I will close this issue though.

closed 2019-04-20 by CorvetteCole