MarlinFirmware/Marlin #13759: Z-Offset don´t change 0 position

arnold-b, 2019-04-18:
Hello, 

if i change Z-offset with M851, M500 or in the sourcecode it has no influence to the 0 position. 

It means my nozzel crash always in the bed. Even i make the offset more high, nothing will be changed. 

If i do G29 with diffent Z-Offset then the result will be related to the new z-offset, but Z0 still crash.

In my sample the nozel touch the Bed at Z 3.01 what ever i do

I´m using Bugfix 2.0 and  Z - Max Endstop. No Z-Min

RudyBenner, 2019-04-19:
Use Zmin for your endstop.
Do the self test.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle