MarlinFirmware/Marlin #13761: [BUG] [bugfix-2.0.x] Dual Z setup, Z2/E1 doesn't move

JustGand, 2019-04-19:
### Description

I have a MSK BASE V1.4 board, RAMPS 1.4 compatible.
When i activate the Z_DUAL_STEPPER_DRIVERS option, the second Z motor does not move.

### Steps to Reproduce

1. #define MOTHERBOARD BOARD_MKS_BASE_14 in Configuration.h
2. #define Z_DUAL_STEPPER_DRIVERS in Configuration_adv.h
3. Compile, Upload
4. Try to move Z

**Expected behavior:** Both Z motors should run at the same speed and direction

**Actual behavior:** Z1 motor turns as expected, Z2 motor does not move at all

#### Additional Information

When I lock all the steppers with M17 command, Z2 is locked ; when I unlock E1 (M18 E1) Z2 motor is set free.

Here are my Configuration.h and Configuration_adv.h
[conf_n_conf_adv.zip](https://github.com/MarlinFirmware/Marlin/files/3098748/conf_n_conf_adv.zip)

Bob-the-Kuhn, 2019-04-22:
I just dumped your configuration files to a 2560/RAMPS system.  The two Z motors worked as expected.

Please provide a picture of the controller board with all the steppers unplugged except for the two Z steppers. The two Z steppers should be plugged into the Z and E1 slots.

Bob-the-Kuhn, 2019-04-22:
Also, please use the following file.  It has the comments restored to the original.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3102248/Configuration.zip)

JustGand, 2019-04-22:
A very weird thing happened : I used the Configuration.h from your previous message.
Z2/E1 was working, but Z1 didn't ...

Then it occurred to me the bed heat sensor was disabled ; i modified Configuration.h (#define TEMP_SENSOR_BED 1), reuploaded it, and then ... Z1 and Z2 are working flawlessly (as are the other motors)

[image]

Bob-the-Kuhn, 2019-04-22:
Definitely weird.  The presence/absence of the bed temperature sensor should not affect if/how the steppers operate.

Glad to hear you are fully functional.

Please close this issue if we're done.

JustGand, 2019-04-22:
I was waiting for your answer in case you had any insight ...
I'm closing it know, but yeah, very weird !

closed 2019-04-22 by JustGand

Earluke, 2019-05-16:
exactly the same board and Problem - but activating the bed temp sensor did not help