MarlinFirmware/Marlin #13763: [BUG] G29 J probes outside of bed area

alexxy, 2019-04-19:
### Description
If i do  second print with UBL (i use G29 J to apply UBL data). By second i mean second print after printer switched on. G29 J probes outside of bed area

Configuration.h has

// Direction of endstops when homing; 1=MAX, -1=MIN
// :[-1,1]
#define X_HOME_DIR -1
#define Y_HOME_DIR  1
#define Z_HOME_DIR  1

// @section machine

// The size of the print bed
#define X_BED_SIZE 380
#define Y_BED_SIZE 380

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS -35
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 445

So my X endstop are outside of bed area (setup similar to this #12634)

Configuration_adv.h

#if EITHER(MESH_BED_LEVELING, AUTO_BED_LEVELING_UBL)
  // Override the mesh area if the automatic (max) area is too large
  #define MESH_MIN_X MESH_INSET
  #define MESH_MIN_Y MESH_INSET
  #define MESH_MAX_X X_BED_SIZE - (MESH_INSET)
  #define MESH_MAX_Y Y_BED_SIZE - (MESH_INSET)
#endif

Usualy it tryes to probe somehere X-4.. or someting

### Steps to Reproduce
1. Turn on printer. Do one print with UBL
2. Do second print with UBL

**Expected behavior:** [What you expect to happen]
Second print should work fine.
**Actual behavior:** [What actually happens]
Second print will fail because G29 J will start probbing outside of bed
#### Additional Information
[config.zip](https://github.com/MarlinFirmware/Marlin/files/3098879/config.zip)
[image]

Roxy-3D, 2019-04-19:
/**
 * Points to probe for all 3-point Leveling procedures.
 * Override if the automatically selected points are inadequate.
 */
#if EITHER(AUTO_BED_LEVELING_3POINT, AUTO_BED_LEVELING_UBL)
  //#define PROBE_PT_1_X 15
  //#define PROBE_PT_1_Y 180
  //#define PROBE_PT_2_X 15
  //#define PROBE_PT_2_Y 20
  //#define PROBE_PT_3_X 170
  //#define PROBE_PT_3_Y 20
#endif

gloomyandy, 2019-04-19:
Looks like you have some sort of dual nozzle setup. I suspect that the problem may be related to that. Do you see any difference if your first print only uses one nozzle rather than two?

alexxy, 2019-04-19:
@gloomyandy i see no difference. But seems like its cleanly a bug. Since first print works fine, while second always fail (due to issue)

Another possibility that it can be related to miss spi connection to dirvers... 

```
Send: M122
Recv: 		X	Y	Z	E	E1
Recv: Enabled		false	false	false	false	false
Recv: Set current	1200	1200	1400	1200	1200
Recv: RMS current	1164	1164	1370	1164	1164
Recv: MAX current	1641	1641	1932	1641	1641
Recv: Run current	16/31	16/31	19/31	16/31	16/31
Recv: Hold current
Recv: CS actual
Recv: PWM scale
Recv: vsense		0=.310	0=.310	0=.310	0=.310	0=.310
Recv: stealthChop
Recv: msteps		16	16	16	16	16
Recv: tstep
Recv: pwm
Recv: threshold
Recv: [mm/s]
Recv: OT prewarn
Recv: off time		0	0	0	0	0
Recv: blank time	24	24	24	24	24
Recv: hysteresis
Recv: -end		-1	-1	-1	-1	-1
Recv: -start		1	1	1	1	1
Recv: Stallguard thrs	0	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	E	E1
Recv: stst
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:
Recv: 		X	0x00:00:00:00	 Bad response!
Recv: 		Y	0x00:00:00:00	 Bad response!
Recv: 		Z	0x00:00:00:00	 Bad response!
Recv: 		E	0x00:00:00:00	 Bad response!
Recv: 		E1	0x00:00:00:00	 Bad response!
Recv: 
Recv: 
Recv: Testing X connection... Error: All LOW
Recv: Testing Y connection... Error: All LOW
Recv: Testing Z connection... Error: All LOW
Recv: Testing E connection... Error: All LOW
Recv: Testing E1 connection... Error: All LOW
Recv: ok

```

alexxy, 2019-04-19:
> /**
>
> * Points to probe for all 3-point Leveling procedures.
> * Override if the automatically selected points are inadequate.
>   */
>   #if EITHER(AUTO_BED_LEVELING_3POINT, AUTO_BED_LEVELING_UBL) [...]

Ok. But why it works when it started first time? =)

alexxy, 2019-04-19:
Start G-code

```
M107
M190 S115 ; set bed temperature and wait for it to be reached
M109 S245 T1 ; set temperature and wait for it to be reached
G29 D; disable UBL
G28; home all 
;;; Measure UBL
;G29 P1; enable UBL
;G29 P3; Fill missing
;G29 S1; Save UBL to EEPROM to slot 1
;;; Load Measured UBL
G29 L1; Load UBL from slot 1
G29 J; probe 3 poitns to land mesh
;;;
G29 A; Activate UBL
G1 Z5 F5000; lift nozzle
; Filament gcode
```

gloomyandy, 2019-04-19:
Is there any chance that on the second print you have a different tool selected compared with the first print?

alexxy, 2019-04-19:
> Is there any chance that on the second print you have a different tool selected compared with the first print?

No. Same gcode file =) So its realy a bug.

gloomyandy, 2019-04-19:
Actually I was thinking the perhaps at startup tool 1 was selected but that your print job results in say tool 2 being selected at the end of the job, so that when you run the file again the current tool when you run G29 is now tool 2, if you see what I mean!

But in any case I'm out of ideas. All I know is that I've never seen this issue and I use G29 J at the start of all my prints and use UBL. I also have a min pos that is -ve (like you). I will often print multiple times in a session, with no problem. So I think there is something in your setup that is triggering the issue, my money is still on it being the multiple nozzles!

Just a thought can you run a single nozzle firmware on your setup? Might be worth trying that to see if the problem still exists?

alexxy, 2019-04-20:
@gloomyandy Arent g29 should respect Z_PROBE offset?

```
#define X_PROBE_OFFSET_FROM_EXTRUDER    2   // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER   27   // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -0.9   // Z offset: -below +above  [the nozzle]
```

And second head has 
```
#define HOTEND_OFFSET_X {0.0, 35.00} // (mm) relative X-offset for each nozzle
#define HOTEND_OFFSET_Y {0.0, 1.00}  // (mm) relative Y-offset for each nozzle
```

So bug may be related to Z_PROBE offset (i see that it uses mesh and inset parameters) So it may be substracted twice from current possition?

I can try single nozzle setup, or force switching to nozzle 1 before probing (but it suppose that it should be automaticaly switch to Z_Probe coordinate system)

gloomyandy, 2019-04-20:
Yes I'm sure it is supposed to work with a dual nozzle setup, but I'm just trying to narrow it down. G29 usually works fine with probe offsets (at least with a single nozzle).

alexxy, 2019-04-22:
Seems like workaround is forcing first nozzle in gcode before probbing...

```
M107
M190 S115 ; set bed temperature and wait for it to be reached
M109 S245 T1 ; set temperature and wait for it to be reached
T0
G29 D; disable UBL
G28; home all 
;;; Measure UBL
;G29 P1; enable UBL
;G29 P3; Fill missing
;G29 S1; Save UBL to EEPROM to slot 1
;;; Load Measured UBL
G29 L1; Load UBL from slot 1
G29 J; probe 3 poitns to land mesh
;;;
G29 A; Activate UBL
G1 Z5 F5000; lift nozzle
```

alexxy, 2019-04-25:
@gloomyandy any ideas?

gloomyandy, 2019-04-25:
Looks like you have identified the problem (and a work around), but I'm not familiar with that code so I can't help come up with a fix for it I'm afraid. Perhaps @Roxy-3D or @thinkyhead will have some suggestions?

Roxy-3D, 2019-04-28:
I'm wondering if the 3-point mesh tilting should be re-written...

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-09-24:
@alexxy is the issue still present?

alexxy, 2019-09-24:
Yes its still present. If you're tool is not T0 then you have an issue.

Workaround is to switch to T0 when do probing

InsanityAutomation, 2019-10-15:
@boelle can you set the title to bug confirmed and add when T0 not active with hotend offsets applied

boelle, 2019-10-15:
like that?

InsanityAutomation, 2019-10-15:
Yup! I wish I had seen this before I worked all around it with M851XY lol oh well ill keep it in mind next time im there...

boelle, 2019-11-03:
@alexxy many updates have passed, is this still an issue on latest 2.0.x?

alexxy, 2019-11-07:
Bug still here, so i still use workarount

boelle, 2019-12-02:
@alexxy is the workaround still needed?

boelle, 2019-12-07:
@alexxy since 2.0 was just released a few days ago has this changed this issue at all?

boelle, 2020-01-24:
@alexxy when you get time please have a retest

and as bugfix 2.0.x is updated daily please retest say every 14 days or so

alexxy, 2020-01-30:
Bug still present on bugfix-2.0.x

alexxy, 2020-01-30:
I think solution is to force T0 for any G28 g29 commands and restore old tool after that

boelle, 2020-02-02:
@alexxy doesit work if you switch to T0 before G29 and then switch back after G29 ?

@thinkyhead do it make sense to you?

alexxy, 2020-02-03:
@boelle Yes it works. Thats why i suggested fix.

InsanityAutomation, 2020-02-03:
@alexxy I dropped the ball and forgot about it, sorry! Try out the PR I just opened.

thinkyhead, 2020-02-04:
I will have a crack at this too. The most comprehensive solution is to consider the current tool's hotend offset when reckoning where to move the nozzle-relative "current position" during probing. All current methods that move to a "probe relative" position will need to be updated with this consideration.

InsanityAutomation, 2020-02-04:
@thinkyhead it shouldn't be an issue if we ensure T0 is active as the open PR does. Not many places that's missed nowadays.

thinkyhead, 2020-02-04:
That is a fine workaround to prevent the issue, but switching tools is not strictly necessary for many setups, such as those with switching nozzles and so on. Inevitably this will be given a default behavior (only change tools if the extruder type demands it) and a configuration option to specify which tool must be active for probing.

InsanityAutomation, 2020-02-04:
More following current standard convention rather than a refactor at this point.

Regardless at some point better control would be good. Even if we don't need to switch tools (single nozzle, mixing excluded) any offset needs to be unapplied from the position. Or the offset of the "probe tool" applied. Probably best in the probe deploy / stow method rather than probe at point so it's less duplicated.

thinkyhead, 2020-02-04:
> More following current standard convention rather than a refactor at this point.

_In the beginner's mind there are many possibilities, in the expert's mind there are few._

There is no convention except our ad hoc. And this is, in fact, one of the easier problems to solve. Every instance of `probe.offset_xy` indicates that the probe position is involved in the move. The offset to the probe simply shifts over depending on the selected tool, so it's just a matter adding one extra level of indirection for that offset when there are multiple hotends.

InsanityAutomation, 2020-02-09:
@boelle as pr is merged and op has not responded to verify, please go ahead and close this for now. If we missed something, he can reopen as needed.

closed 2020-02-09 by boelle