MarlinFirmware/Marlin #13764: BLTouch V3 prints in air, (Z-OFFSET and Z_PROBE_OFFSET settings are correct)

sgsalman, 2019-04-19:
Hello I just set my setting show in this video (https://www.youtube.com/watch?v=sOFxalLOZOI&t=469s) (second method)
I set the Z-offset(-2) with using A4 paper. I put a G29 command to slicer. Also I set the Z_PROBE_OFFSET_FROM_EXTRUDER as 1.1(bltouch pin's is above 1.1m from extruder). When I start prints meshing(G29) is fine work. But printer prints in air. I couldn't recognize what is the problem. Im adding my conf.h and conf_adv.h files. 

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3099016/Configuration.zip)

RudyBenner, 2019-04-19:
Which version are you using?

sgsalman, 2019-04-19:
Mainboard 1.1.4 
Firmware 1.1.9 bugfix
Bltouch v3 pre april

RudyBenner, 2019-04-19:
bugfix1.0.x then?

sgsalman, 2019-04-19:
I downloaded it from very soo

RudyBenner, 2019-04-19:
Line 738 why is this uncommented? You have 728 uncommented.

[image]

No idea about 727

Is there a reason for line 731.
The pin file says that 27 is for the beeper. 
Line 1928 is where servo0 is usually selected.
I am probably missing something.

[image]

[image]

[image]

Anyway, have a look at the BLTouch v3 thing first. There could be more, but I don't want to spoil your anguish.

RudyBenner, 2019-04-19:
Remember to do M502 followed by M500 after flashing your firmware.
G29 generated the mesh. M420 S1 activates the mesh. Put M420 S1 in your start script AFTER any instance of G28 if there is one. When it is printing, you can feel the Z stepper moving very slightly as the gantry moves.

reloxx13, 2019-04-20:
for ender 3 u cap the beeper line and use its pin for the bltouch. the ender3 board has no more easy accessable pins.

reloxx13, 2019-04-20:
use m851 to set ur z offset on not teaching tech method of babystepping it.

> M851 Z0.00 ;set z offset to 0
> m500 ;save
> m501 ;load
> 
> g28 ;home
> G90 ;use absolute mode
> M211 S0 ;disable software endstops
> G1 Z0 ;z on real 0
> 
> now put in ur paper and lower z till it rubs the paper
> !!!this is absolut, not relative now - so do small steps!!!
> G1 Z-0.25 ;absolut -0.25mm from 0
> G1 Z-1.00 ;absolut -1mm from 0
> and so on (eg. i´m at -2.25)
> 
> if u got it, save teh z offset
> m851 Z-?.?? ;set ur z-offset
> m211 s1 ;enable software endstops again
> m500 ;save
> m501 ;load
> g28 ;home
> 
> put m420 s1 in ur gcode startcode after g28
> 
> ull still need to finetune it by adjusting it after printing a first layer or skirt. if the skirt looks bad, adjust it by +/- 0,01mm and try again till u find the best value.
> 
> use at own risk, this is how i do it

sgsalman, 2019-04-27:
@reloxx13 I do same things.... 

I perfectly set Z-offset but when print starts, extruder is still above the bed :(

 Is it about extruder offset?
 Can I solve this problem with adding washers btw bltouch and bltouch mount(original kit mount)?

reloxx13, 2019-04-27:
pls post ur gcode startcode

sgsalman, 2019-04-27:
; Ender 3 Custom Start G-code
G28 ; Home all axis
G29 (I ALSO TRIED M420 S1 BUT IT DOESNT WORK)
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
; End of custom start GCode

sgsalman, 2019-04-30:
no help :(

phuxman, 2019-05-01:
Please check during mesh creation if the probe stows sometimes directly after deploy and creates a wrong measurement (too high). I had to increase the probe delay to 500ms so the probe has time to redeploy when this occurs.

Before I did this my printer was also printing in air due to the wrong mesh.

Roish, 2019-05-23:
Having the same issue, the probe will randomly stows directly after deploy. 
Did you increase DELAY_BEFORE_PROBING  to 500?

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

InsanityAutomation, 2019-07-23:
Please note the BLTouch code in 2.0.x Bugfix is far more robust than 1.1.9 so id suggest retrying there and reopen this if the issue persists.