MarlinFirmware/Marlin #13765: [BUG] [bugfix-2.0-x] Bad paths in HAL_TEENSY31_32/HAL.h

MitchBradley, 2019-04-19:
The file referenced in the subject line contains the following lines with incorrect paths:

```
#include "../math_32bit.h"
#include "../HAL_SPI.h"
```
By changing them to 

```
#include "../shared/math_32bit.h"
#include "../shared/HAL_SPI.h"
```
and including a teensy31 [env] section in platformio.ini,
```
#
# Teensy 3.1 / 3.2 (ARM Cortex-M4)
#
[env:teensy31]
platform      = teensy
framework     = arduino
board         = teensy31
build_flags   = ${common.build_flags}
lib_deps      = ${common.lib_deps}
lib_ignore    = Adafruit NeoPixel
src_filter    = ${common.default_src_filter} +<src/HAL/HAL_TEENSY31_32>
monitor_speed = 250000
```
 I got a clean build for BOARD_TEENSY31_32.  The build seems to work to the extent that I can talk to it serially, send GCode and get position reports, but I haven't yet attached any motors or other peripherals.

RaabenFuchs, 2019-07-13:
hi, very nice, i was looking for that! how is your progress so far?

i now have a build ,too. with a teensy 3.0
and i wondered about the pinout, this is all the same on 3.1 and 3.1...

but i dont have a board yet, could you tell, **Is it working for you? What pinout do you use?**

maybe we could arange those pins and do a pull. what do you think? what is it to consider for which carrier board

MitchBradley, 2019-07-13:
Sorry, I have no new information.  I am not planning on hooking up the Teensy4.0/Marlin controller to a machine.  I use it only for serial protocol testing.  I do development and maintenance on CNCjs and I needed to regression-test CNCjs' Marlin protocol support.

RaabenFuchs, 2019-07-13:
hmm nice to know that the serial will work. thank you for your quick response

boelle, 2019-07-21:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

sjasonsmith, 2019-09-21:
It looks like this suggested change was already included in commit [e85eca2630db50574f3829aa9d58237d7562d38f](https://github.com/MarlinFirmware/Marlin/commit/e85eca2630db50574f3829aa9d58237d7562d38f)

closed 2019-09-21 by MitchBradley