MarlinFirmware/Marlin #13770: [BUG] 2.0 bugfix Compiler error TEMP_SENSOR_BED 1

davidhuwyler, 2019-04-20:
### Description
Version Marlin 2.0 Bugfix 20.04.2019.
When I define TEMP_SENSOR_BED with something other than 0 (ex. 1), then 
there are a lot of Compiler errors (see Additional Information)

### Steps to Reproduce
LPC1768 enviroment (with Bigtreetech SKR 1.3 Board)

1. LPC1768 enviroment
2. #define TEMP_SENSOR_BED 1 (in Configuration.h)

#### Additional Information
PlatformIO output:
[TEMP_SENSOR_BED Error.txt](https://github.com/MarlinFirmware/Marlin/files/3100062/TEMP_SENSOR_BED.Error.txt)

ArturNadolski, 2019-04-20:
I got same error.

Patag, 2019-04-20:
Same error here.

Idorobots, 2019-04-20:
A fix for that is here: https://github.com/MarlinFirmware/Marlin/pull/13768
You can checkout that branch before it get's merged.

davidhuwyler, 2019-04-20:
I installed the bugfix. now as soon as i start the controller, a bed thermal runnaway error kicks in

Idorobots, 2019-04-20:
@davidhuwyler  Do you have the thermistor plugged into the board? If you do not it probably reads a very high temperature and kills the printer to avoid any potential damage.

davidhuwyler, 2019-04-20:
yes i actually did plug in a thermistor. did you have no problem with thermal runaway?

Idorobots, 2019-04-20:
I don't have a SKR 1.3 board to test on, but it seems to be working for me on other boards. Perhaps SKR pin definitions aren't mapping the bed thermistor to the connector you expect? You could also try to compile an older version of Marlin, say https://github.com/MarlinFirmware/Marlin/commit/4ac28ba6b367e0c75b62fe51b2b8be334dad18ea should compile fine and it doesn't include the recent thermal protection changes.

seangould, 2019-04-20:
Just as a confirmation, I got a thermal runaway error immediately after booting with the latest 2.0 bugfix branch. This is on a Re-Arm (LPC1768). Reverting back to an earlier release of the 2.0 branch resolves the thermal runaway error on boot.

davidhuwyler, 2019-04-21:
yes i can confirm that too. Latest 2.0 bugfix results in immediat bed thermal runaway error.  reverted back to 4ac28ba and all is fine.

Idorobots, 2019-04-21:
Hey guys, could you please attach your Configuration.h and Configuration_adv.h? I'll try to investigate the problem.

davidhuwyler, 2019-04-21:
Ok configuration file is here:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3101008/Configuration.zip)

No changes made in the Configuration_adv.h file

Idorobots, 2019-04-21:
Yup, I'm able to reproduce the issue on a ReARM board with the included Configuration.h. Curiously enough, this doesn't happen on an ESP32 target.

Idorobots, 2019-04-21:
Looks like the low pass filter in the LPC1768 HAL is causing this issue. I've pushed a fix for this in https://github.com/MarlinFirmware/Marlin/pull/13778, could you guys verify that it works on your boards?

seangould, 2019-04-21:
Just checked, and still getting the same error on boot

Idorobots, 2019-04-21:
@seangould Could you try reducing `ADC_LOWPASS_K_VALUE` even further, or straight up disabling it by setting it to 0? In my case a 4 made it work OK, while at 5 it does error out. Also worth trying is running `platformio run -e LPC1768 -t clean` to clean-build the project.

doggyfan, 2019-04-21:
I presume the low pass filter value is initialized to '0' at boot up ?

Why not make it a much faster LPF at boot up, then switch it to normal after it settles (say after 500ms), during which time you ignore it's output when it comes to the thermal runaway code. That's what I tend too do.

Or probably much more simplier, initialize it's value to 25C, it'll then slide from there to where ever the thermistor is at. This way you can make the LPF's response as slow as you want .. problem solved ?

Idorobots, 2019-04-21:
@doggyfan I've added a short grace period for the readings to stabilize, which by default is disabled except for LPC176* targets, to #13778, is that what you had in mind?

doggyfan, 2019-04-21:
Yes that should work OK Idorobots.

I too had a thermal runaway problem on the bed immediately at bootup .. I have the LPC1768 SKR 1.3 board, but I'm 2 or 3 published builds away from the current one.

davidhuwyler, 2019-04-22:
@Idorobots issue seems to be fixed with the grace period. Thank for investigation & fixing!

closed 2019-04-22 by davidhuwyler

seangould, 2019-04-22:
@Idorobots - lowering the ADC_LOWPASS_K_VALUE to 4 eliminates the thermal runaway on the bed here on my re-arm as well

Idorobots, 2019-04-22:
@seangould You might wanna try out the latest bugfix-2.0 without any extra modifications, as a fix for this issue has been merged yesterday.

seangould, 2019-04-22:
@Idorobots - I did this morning, still had the thermal runaway, lowerered the ADC_LOWPASS_K_VALUE and that resolved the issue here.