MarlinFirmware/Marlin #13771: [BUG] Fysetc Mini 12864 Display goes blank/inoperative when inserting SD card

bipsendk, 2019-04-20:
### Description

Fysetc F6 v1.3 board with Fysetc Mini 12864 display.
Applied patch https://github.com/MarlinFirmware/Marlin/pull/13768
When inserting a SD card in the reader on the display, the display goes blank - only a reset will recover from the problem.

### Steps to Reproduce

1. Compile 2.0-bugfix with patch Applied patch https://github.com/MarlinFirmware/Marlin/pull/13768 applied
2. Upload to Fysetc F6 v1.3 board witrh a Fysetc Mini 12864 display attached
3. Power up board - wait for board to be ready
4. Insert SDcard in reader on LCD board

**Expected behavior:** [What you expect to happen]

Display should write "SD-card inserted" (the usual status message)

**Actual behavior:** [What actually happens]

Display goes blank. Removing the SDcard does not recover from the situation.

#### Additional Information

[config.zip](https://github.com/MarlinFirmware/Marlin/files/3100254/config.zip)

bipsendk, 2019-04-20:
Tried to check/change the display contrast settings after inserting the SDcard. The contrast value does not seem to change when the card is inserted - and changing the contrast value does not affect the display.

Bob-the-Kuhn, 2019-04-20:
On my 2560/RAMPS system I'm also seeing the display blank when inserting or removing the SD card but it always comes back to normal about 1-2 seconds later.

On my Due/RAMPS_FD_V1 system there is no blanking.

I don't see any issues with the signals to the LCD when the blanking happens.

This LCD is turning into a real pain.

Bob-the-Kuhn, 2019-04-20:
Try commenting out `#define LCD_RESET_PIN  23` in pins_FYSETC_F6_13.h.

That changed the "going blank" to "display paused" on my system.

bipsendk, 2019-04-21:
If commenting out LCD_RESET_PIN, the display is not properly initialized on the F6 board - so the display is just "black" (no text/graphics) when powered on...

Bob-the-Kuhn, 2019-04-21:
Same behavior on the SKR?

Do you have  2.0 or 2.1 LCD?

Bob-the-Kuhn, 2019-04-21:
Where did you get  your LCD?  I need to get one.

bipsendk, 2019-04-21:
2.0 LCD (unfortunately I do not have a 2.1 to test with as well).
I bought model B from Aliexpress - https://www.aliexpress.com/item/MINI12864-LCD-Display-Screen-mini-12864-Smart-Display-128x64-5V-Supports-Marlin-DIY-With-SD-Card/32862853197.html?spm=a2g0s.9042311.0.0.27424c4dbXefyq

Bob-the-Kuhn, 2019-04-21:
I've been playing with the reset and looking at the code.  

I expect that if you click the encoder the display will come up.

As best I can tell the display goes blank when both the LCD and SD card selects are active at the same time.  Definitely scratching my head on how that can happen.

Bob-the-Kuhn, 2019-04-22:
Scott just merged the updated Fysetc LCD code.  Please see if that fixes your problem.

bipsendk, 2019-04-22:
Just tested it with fresh code - it is working now (display is blank for a couple of 100msecs when card is inserted - but comes back again pretty quickly).

Bob-the-Kuhn, 2019-04-22:
That's expected.  It's the software resetting the LCD and reading the SD card.

closed 2019-04-22 by Bob-the-Kuhn

Tomza86, 2020-05-12:
Hi, 
I have exact problem. What you did to make it work? 
Thanks