MarlinFirmware/Marlin #13772: UBL Z offset not correct on extruder 2 with  dual extruder

jayholweger, 2019-04-20:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

jayholweger, 2019-04-20:
When using UBL the z offset is based on the native (X,Y) position of the carriage.  When printing using dual extruders on a single carriage the Z offset for the second extruder is incorrect.  My extruder1 offset is 38,0,0.  I have a print that alternates the extuder every layer.  I can tell that the z movements are not the same for extruder0 and extruder1 as they print over the same surface.  

From planner.cpp it looks like the planner is not looking for the extruder offset.  In fact it looks like all of the bed leveling does not look at extruder offsets. 

          #elif ENABLED(AUTO_BED_LEVELING_UBL)
            fade_scaling_factor ? fade_scaling_factor * ubl.get_z_correction(raw[X_AXIS], raw[Y_AXIS]) : 0.0
          #elif ENABLED(AUTO_BED_LEVELING_BILINEAR)
            fade_scaling_factor ? fade_scaling_factor * bilinear_z_offset(raw) : 0.0
          #endif

I certainly may be over simplifying the process since I am not very familiar with this firmware.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

InsanityAutomation, 2019-07-23:
Offset position is behind whats being seen there and has already been applied. XY position reports active tool position and thats what leveling needs to compensate.