MarlinFirmware/Marlin #13773: Marlin  1.1.x  bugfix  and Filement width sensor LCD shows 578 %

bogaskilop, 2019-04-20:
i purchas e filement widht sensor from tindie  and checked it befoure adding to printer  and all measurement was quite right on  multimeter   when i put  filement 1.76 i get  1.76 V on multimeter  but when i configured it in marlin and connected  ( 5V  is supplied form separate power suplly to avoid wrong readings  as sugested by manufacturer)

configuration.h 

#define DEFAULT_NOMINAL_FILAMENT_DIA 1.75

configuration_adv.h 
#define FILAMENT_WIDTH_SENSOR

#if ENABLED(FILAMENT_WIDTH_SENSOR)
  #define FILAMENT_SENSOR_EXTRUDER_NUM 0    // Index of the extruder that has the filament sensor. :[0,1,2,3,4]
  #define MEASUREMENT_DELAY_CM        13.5 // 14    // (cm) The distance from the filament sensor to the melting chamber

  #define FILWIDTH_ERROR_MARGIN        1.0  // (mm) If a measurement differs too much from nominal width ignore it
  #define MAX_MEASUREMENT_DELAY       20    // (bytes) Buffer size for stored measurements (1 byte per cm). Must be larger than MEASUREMENT_DELAY_CM.

  #define DEFAULT_MEASURED_FILAMENT_DIA DEFAULT_NOMINAL_FILAMENT_DIA // Set measured to nominal initially

  // Display filament width on the LCD status line. Status messages will expire after 5 seconds.
  #define FILAMENT_LCD_DISPLAY
#endif

I also needed to turn on volumetrics
//#define NO_VOLUMETRICS

#if DISABLED(NO_VOLUMETRICS)
  /**
   * Volumetric extrusion default state
   * Activate to make volumetric extrusion the default method,
   * with DEFAULT_NOMINAL_FILAMENT_DIA as the default diameter.
   *
   * M200 D0 to disable, M200 Dn to set a new diameter.
   */
  #define VOLUMETRIC_DEFAULT_ON
#endif

and   now  i put  filament  that has  1.76 mm diameter and that what i get on multimeter  
but  on  LCD i get  578 %   and when execute M407 :

Send: M407
Recv: Filament dia (measured mm):4.05
Recv: ok

bogaskilop, 2019-04-20:
it looks like it multiply   actuall filament width  by 2.85

bogaskilop, 2019-04-25:
is there any one who can check if in new bugfix  a filament sensor is working ?

bogaskilop, 2019-05-02:
@thinkyhead  can you take a look on this ?  thank you

bogaskilop, 2019-05-15:
i will check this but i think i just connected 5v and negative  to sensor  as  creator wrote  that it is better to power sensor from separate power supply  so i connected +5 and negative from psu to sensor   and only signal goes to mks board

InsanityAutomation, 2019-05-15:
If the - sides arnt connected the board will not be able to recognize the signal.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle