MarlinFirmware/Marlin #13774: No fan spin with extruder fan or parts cooling fan on Bugfix-2.0 and MKS SBASE

bootstrappy, 2019-04-21:
### Description

As the title says, I have Marlin 2.0 flashed onto two different MKS SBASE 1.3 boards.  Motion controls are fine.  Bed and extruder heaters work fine. All temp sensors work. But there is ZERO fan spin when an m106 command is entered and it simply doesn't matter when entered or what temperature anything is at.   No fans spin to any degree. These are known good fans that were first function tested by being individually plugged into a PC PSU. Again, this behavior is identical on both of my boards.

### Steps to Reproduce

1. Unit is fully powered.
2. M106 is entered or extruder is heated to over 50C.

**Expected behavior:** 
Fans spin.
**Actual behavior:**
No fan spin.

Included in the zip are the three files that I have changed -configuration.h, configuration.adv.h, and pins_mks_sbase.h

[Config_files.zip](https://github.com/MarlinFirmware/Marlin/files/3100835/Config_files.zip)

Ketchu13, 2019-04-25:
You don't have defined the pin for the extruder auto fan: `#define E0_AUTO_FAN_PIN -1`
And maybe you have to try to undef  `#define FAN_SOFT_PWM`

DerAndere1, 2019-04-25:
According to issue #13389 you should start with a very recent bugfix-2.0.x version (less than 29 days since download from github). There was some time when FAN_SOFT_PWM was broken obviously.

bootstrappy, 2019-04-27:
I had already tried E0_AUTO_FAN_PIN and FAN_SOFT_PWM a couple of times and this problem has been consistent for two months now, March and April.  I wish that I had an older version to try, but March was when I first heard that Marlin2.0 even existed and couldn't dump Smoothieware fast enough. I'll try again and see if this time is different. Let you know tomorrow.

gloomyandy, 2019-04-27:
This configuration was working fine for me using the E1 heater mosfet to control the fan on the extruder....
``` C
#define E0_AUTO_FAN_PIN HEATER_1_PIN
#define E1_AUTO_FAN_PIN -1
#define E2_AUTO_FAN_PIN -1
#define E3_AUTO_FAN_PIN -1
#define E4_AUTO_FAN_PIN -1
#define E5_AUTO_FAN_PIN -1
#define CHAMBER_AUTO_FAN_PIN -1
#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED 255 
```

bootstrappy, 2019-04-28:
Downloaded today's version of the firmware, extracted it, set up all the files all over again, tried it with FAN_SOFT_PWM enabled and disabled, tried #define E0_AUTO_FAN_PIN HEATER_1_PIN( and fan physically moved to 2_06) and E0_AUTO_FAN_PIN FAN_PIN from the pins.h and still nothing. Again, this is on two different boards, so it's software. We're either still missing something, or the firmware slightly broken in regard to MKS_SBASE boards.

DerAndere1, 2019-06-04:
Someone in the facebook group reported a very similar bug: With an SBASE 1.3 board he could start the extruder fan only via the display menu. The part cooling fan did not work at all for him.

bootstrappy, 2019-06-04:
Thanks.  I give that a try.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

SergioSV96, 2019-08-27:
I am having the same problem, I can start the fan with the LCD display, but it won't start automatically when printing.

I am using Anycubic Kossel Pulley, with a Trigorilla Board.

DerAndere1, 2019-09-19:
There is a new duplicate of this
issue: https://github.com/MarlinFirmware/Marlin/issues/15199?_pjax=%23js-repo-pjax-container with solutions for the MKS SBASE 1.3 .

boelle, 2019-10-12:
@bootstrappy still having issues?

closed 2019-10-24 by boelle