MarlinFirmware/Marlin #13775: Extruder Stepper not working properly

Aurawin, 2019-04-21:
When I try hot/cold extrude my stepper motor hums and sputters.  Interestingly, the motor works with expected torq in reverse.  At first I thought it was my driver DRV8825 but after testing it - this problem is with the firmware.  If I rollback to 1.9 I can extrude just fine.  Virtually the same configs with the exception of BLTouch 3.0.

### Steps to Reproduce

1. M302 P1 
2. Extrude forward (stepper sputters)
3. Extrude reverse (stepper responds as expected)

**Expected behavior:** [What you expect to happen]

I expect the forward rotation to work as the reverse rotation

**Actual behavior:** [What actually happens]

The extruder (E0) will only rotate flawlessly in reverse

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3100891/Marlin.zip)

FelixHauser, 2019-04-22:
Do you have Linear Advance enabled? If so, disable it and try again. I had also problems with the extruder motor and it was LA at fault

marcio-ao, 2019-04-22:
Are you guys using 32-bit boards? If so, look for the workaround here:

https://github.com/MarlinFirmware/Marlin/issues/13040

FelixHauser, 2019-04-22:
> Are you guys using 32-bit boards? If so, look for the workaround here:
> 
> #13040

I assumed he/she is, since it was mentioned rolling back to 1.1.9... that's why I suggested to disable LA on a first place

marcio-ao, 2019-04-22:
@FelixHauser: It is not necessary to disable LA. The workaround is to set MINIMUM_STEPPER_PULSE to 1.

FelixHauser, 2019-04-22:
Yes, yes, I know...

Aurawin, 2019-04-25:
The board I'm using is a Tango (Rumba pins)
The printer is a corexy with dual z (TRONXY X5S)

The Z2 stepper is attached to the E1 port / pins with NO problems

I have extra Extruder ports so I remapped the pins from E2->E0.  

If I set INVERT_E0_DIR to E0 to true the extruder rotates during the E-# only again, even though the actual motor IS reversed direction as compared to before.  

I have since tested with another stepper motor.  Same result.

I will say that the pulse pattern results in sputter at first then the signal is provided a bit more normal towards the end of the pulse process.  A pulse process in the E-# direction lasts much longer than the E# direction.

The problem is exacerbated when I issue larger movements.  It is hardly noticeable when the motions are ~1mm

setting MINIMUM_STEPPER_PULSE to 0 or 1 does improve the symptoms but is not a solution.  The signal is only slightly improved not perfect like in the E-# direction.

Aurawin, 2019-04-26:
G0 E30 F200 works fine
G0 E30 F2000 works fine
G0 E30 F20000 reproduces the failure

So looking at Repetier manual moves the gcode is issuing speeds of 18000 on the forward and 1800 on the reverse manual extrusion moves.  

Looking at the firmware settings :

#define DEFAULT_MAX_FEEDRATE          { 300, 300, 25, 10 }
Yields a "working" limit on feed rate that even when Repetier sends a G1 E50 F12000 the firmware limit is low enough to generate valid stepper signal.

#define DEFAULT_MAX_FEEDRATE          { 300, 300, 25, 30 } reduces to an acceptable feedrate
#define DEFAULT_MAX_FEEDRATE          { 300, 300, 25, 60 } is not reducing to an acceptable feedrate

60mm/sec is a valid albeit high feed rate value for an extruder but not a stepper motor.

Shouldn't Marlin have reduced the value before driving the stepper?

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle