MarlinFirmware/Marlin #13777: [BUG] SD upload issue M29 / M290 (Marlin 1.1.x)

MuellerDominik, 2019-04-21:
### Description

When sending the `M290` command during a serial SD card upload, it stops writing to the file.

This issue is present in the `1.1.x` branch, as well as in the `bugfix-1.1.x` branch. It has been fixed in the `bugfix-2.0.x` branch (see further below).

### Steps to Reproduce

1. Clone Marlin
1. Stay in `1.1.x` branch or switch to `bugfix-1.1.x` branch (issue is present in both branches)
1. Enable `SDSUPPORT` in `Configuration.h`
1. Enable `BABYSTEPPING` in `Configuration_adv.h`
1. Build and upload (I've used PlatformIO, but that's not important in this case)
1. Connect SD card (make sure it's initialized with `M21` and no file called `test.gco` exists)

The following example shows the issue (SD card must be properly initialized):
```
M110 N0
N1 M28 /test.gco*68
N2 M290*42
N3 G0 X20*80
N4 M29*28
```

**Expected behavior:** A new file `test.gco` with the following contents:
```
M290
G0 X20
```

**Actual behavior:** A new file `test.gco` with no content.

#### Additional Information

The problem is, that the command `N2 M290*42` (from the example above) actually stops the SD transfer whereas it should write the command `M290` to the file.

This happens because the [following line](https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/Marlin_main.cpp#L15316) doesn't check, if the character following the string `M29` is numeric, when the substring `M29` is found in the command.

I suggest a fix similar to the one applied in the [following commit](https://github.com/MarlinFirmware/Marlin/commit/92ed9a2bb6d90a969f2ceaf0a73ebe3c0cdc4c16) to the `bugfix-2.0.x` branch. Let me know if I should create a PR or if you're going to fix this small issue yourself.

Further information on this issue:

* https://github.com/luc-github/ESP3D/issues/316
* https://github.com/MarlinFirmware/Marlin/pull/13013 (same issue in `bugfix-2.0.x` branch)
* https://github.com/MarlinFirmware/Marlin/issues/11871 (related)

ZIP file containing my [configuration files](https://github.com/MarlinFirmware/Marlin/files/3101090/Configuration_files.zip).

thinkyhead, 2019-05-16:
We won’t be fixing this in 1.1.x and I think it is at least mostly fixed in 2.0.x. Please test 2.0.x and report what is still borked there. Did #13013 not address it fully?

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

MuellerDominik, 2019-07-21:
@thinkyhead no worries, https://github.com/MarlinFirmware/Marlin/pull/13013 completely fixed the issue for the `bugfix-2.0.x` branch (as stated in my initial post above). Since you're not going to fix the bug in the `1.1.x` branch nor in the `bugfix-1.1.x` branch (even though it would be an easy fix) this issue can remain closed.