MarlinFirmware/Marlin #13779: [BUG] Bugfix-2.0 - Z-max endstop not working after a power-up

bipsendk, 2019-04-21:
### Description

In Marlin 2.0-bugfix it seems like the Z-max endstop not is working after a powerup.
If a printer (HEVO) is equipped with Z-Max endstop, and the buildplate is near maximum height - then it is possible to proceed to increase the height, even though the Z-Max has been triggered.

### Steps to Reproduce

1. Establish Z-Max endstop on printer
2. Move Z axis to max-height-10mm
3. Power cycle printer (or reset controller)
4. In LCD panel, try to increase Z axis height by 12 mm

**Expected behavior:** [What you expect to happen]

Z-axis movement would stop at 10mm - as the Z-Max endstop has been triggered at that point (even though no homing has been performed)

**Actual behavior:** [What actually happens]

Z axis increased height by 12mm - even though Z-max endstop was triggered at 10mm

#### Additional Information

Endstops has been tested with M119 - which clearly indicates, that Z-Max endstop has been triggered at 10mm. Even though Z-Max has been triggered, it is still possible to increase Z height...

kAdonis, 2019-04-21:
You didn't post your configs, but I assume you are homing to z-min. Afaik, by default marlin is polling endstops only when homing, so more than one endstop per axis makes no sense. The word "endstop" is misleading, "homing-switch" would be better.
To get your expected behavior you could try to uncomment:
`//#define ENDSTOPS_ALWAYS_ON_DEFAULT`
but you might run into other problems

bipsendk, 2019-04-21:
Yes, homing is done to Z-min (when relevant).
My initial thought was that they acted as "limit switch" as well, so that if it was triggered, no further movement in that direction was possible.
But another issue might pop up - if you have printed to maximum height, powered off the printer, power it on again - and the do home all (with Z safe homing, where it lifts the extruder a certain amount, I cannot remember if the distance can be set) - then you will risk exceeding the max Z height. I am not sure whether the Z-max endstop is enabled/checked during the homing process (if homing to Z-min) - but if not, there is potential for crashing stuff if trying to exceed the Z limit...

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle