MarlinFirmware/Marlin #13783: [BUG] Bugfix-2.0 Bed thermal runaway triggered right after reset - even though disabled

bipsendk, 2019-04-21:
### Description

Took a fresh download of the 2.0-bugfix repository today (approx 20:15 CET)

Compiled for SKR v1.3 with Fysetc 12864 display.

Compiled the code and uploaded to the board - but as soon as the new firmware is booted, the display shows "BED THERMAL RUNAWAY"

This even happens with `THERMAL_PROTECTION_BED` commented out

### Steps to Reproduce

1. Configure as with the supplied config files - set target to LPC1768 in platformio.ini
2. Compile in VScode
3. Upload firmware to SDcard on SKR v1.3 board
4. Reset board

**Expected behavior:**

Board would boot and show the normal info/status screen

**Actual behavior:** [What actually happens]

BED THERMAL RUNAWAY

displayed on LCD
[BedThermalRunaway.zip](https://github.com/MarlinFirmware/Marlin/files/3101497/BedThermalRunaway.zip)

FanDjango, 2019-04-21:
I confirm, happens to me to. But they're on to it methinks: #13770

restoration2699, 2019-04-21:
also if you set hotend temp about 240 degrees and after that reset the board you will get almost the same error, but with hotend thermal protection.
something is wrong in last build.

Idorobots, 2019-04-21:
Hey guys, here's a fix if you'd like to test it: https://github.com/MarlinFirmware/Marlin/pull/13778

bipsendk, 2019-04-24:
Seems to be fixed in latest version I grabbed...

closed 2019-04-24 by bipsendk

spilz87, 2019-04-26:
hello,

i install the last update, included #13778 but I still have this issue on my BIGTREE_SKR_V1_3.
it still not solved

(for information, at 866e2d41dca277a1913a7e505a3d07d6d10f7ce3 i Don't have this issue)

thanks for your help

devolf92, 2019-04-26:
same error after restart with heated hotend. need to restart 3-5 times to skip this error
skr 1.3 with only hotend termistor. default thermal runaway settings

Idorobots, 2019-04-26:
Probably SKR needs a longer grace period. Could you guys try bumping the `THERMAL_PROTECTION_GRACE_PERIOD` in `HAL_LPC1768/HAL.h` to a higher value and try again?

devolf92, 2019-04-27:
I have find out core of my issue
I have no any THERMAL_PROTECTION_GRACE_PERIOD in HAL_LPC1768/HAL.h , so i updated to current marlin beta and for a now everything works fine. can't get a clue how i had half of firmware updated.

except skipping steps in STEALTHCHOP, dunno why

spilz87, 2019-04-27:
Is it only on startup ? 
I think a part of the issue is that my MKR take about 5 secondes to start, and I get directly this thermal runaway. 
So I guess I only need a delay on start, not after ... 
I’m going to test with THERMAL_PROTECTION_GRACE_PERIOD at 5000
I will let you know

spilz87, 2019-04-27:
I just text at 5000, then 2000 and finally 1000, they all works fine, but 500 is not ok

thinkyhead, 2019-04-27:
How about 750?

spilz87, 2019-04-27:
750 is ok

cinealfa, 2019-04-27:
With THERMAL_PROTECTION_GRACE_PERIOD set 800 it doesnt work with MKS-sbase going to try 1000

Brainc0re, 2019-04-28:
I had a fysetc display, but i want to back to the reprap full graphic display.

So i took the latest build (2h ago) made all changes, inserted the SD, pressed the reset button to install the FW and i get the thermal runaway error directly.
If i connect the fysetc display again with the marlin version from 24.04.2019 no error.
So i thought, i take the 24.04 release and change only the display type - but i get the error at the startup.