MarlinFirmware/Marlin #13785: [BUG] Bugfix-2.0 lost 20mm Z-axis with tmc2208

DimchikUA, 2019-04-22:
### Description
Hello!
I have a BOARD_BIGTREE_SKR_V1_3 board for my custom 3Dprinter - coreXY .
Earlier I used the DRV8825 driver with it and everything worked fine. Now I decided to use TMC2208 (uart mode) for the X and Y axis. The Z and E axes continue to use the DRV8825. I configured the firmware and flashed it, everything seemed to work fine until I started the test print.
Axis X, Y and Z uses max direction when run homming. Z max pos = 298mm. After homing Z-axis and G1 Z0, I have a gap of ~ 20mm between the table and the nozzle. When I tell the Z axis to drive 10mm it drives through 9.2-9.3mm.
When I use in firmware the DRV8825 for XY axis everything works fine.
And I do not understand in what matter. I have not made any changes regarding the movement of the Z axis.

### Steps to Reproduce

**1. [First Step] XY axes use drv8825**
```
* Stepper Drivers
...
#define X_DRIVER_TYPE  DRV8825
#define Y_DRIVER_TYPE  DRV8825
#define Z_DRIVER_TYPE  DRV8825
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE A4988
//#define Z3_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE DRV8825
//#define E1_DRIVER_TYPE A4988
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988
```
Everything works well.

**2. [Second Step] Change DRV8825 to TMC2208 for XY Axes**
```
#define X_DRIVER_TYPE  TMC2208
#define Y_DRIVER_TYPE  TMC2208
#define Z_DRIVER_TYPE  DRV8825
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE A4988
//#define Z3_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE DRV8825
//#define E1_DRIVER_TYPE A4988
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988
```
I get the movement of the axis Z about 9.2-9.3 mm instead of 10 given.

**3. [Third Step] I return DRV8825 axes XY**
Z axis is moving correctly.

#### Additional Information

I do not make any other changes, only these two lines where I set the drivers for the X axis and the Y axis.
[Marlin-bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/3102858/Marlin-bugfix-2.0.x.zip)

reloxx13, 2019-04-22:
http://marlinfw.org/docs/configuration/configuration.html#default-steps-per-mm-

DimchikUA, 2019-04-22:
**кeloxx13**, this is not it! How to calculate the number of steps per millimeter, I know. In my firmware, I do not change the number of steps per millimeter for the Z axis. I only change the drivers for the X and Y motors.
I don't want to change the number of steps per mm, because I have a calculated number of 3200, and it works until I start using another driver, and not for this axis, but for others.

kAdonis, 2019-04-22:
There is a known issue with corexy printers with 32bit MCUs and DRV8825 (step pulses too short) #11047
try `#define MINIMUM_STEPPER_PULSE 3`
Are you shure you had no layer shifts with DRV8825 on all axes?

teemuatlut, 2019-04-22:
So which one is correct? Get a measuring device of your choosing and see how everything differs from theoretical to reality.

DimchikUA, 2019-04-22:
Thank you, **kAdonis**. Problem solved using `#define MINIMUM_STEPPER_PULSE 3`
Before that, I used an 8-bit board with the drivers of drv8825. After upgrading to 32-bit, I decided not to rush to install tmc2208, and pre-test everything with 100% working drivers. There were skip steps, of course. The solution was #define MINIMUM_STEPPER_PULSE 2, as written in the comments of the firmware and it worked.
After installing tmc2208 for XY axes, I could not even think about skipping steps along the Z axis. I was confused the accuracy of repetitions. After going back and forth, the table always returned to the same real point. In addition, the extruder moved the thread fairly accurately, I measured.
It seems that the steps were lost strictly with some kind of periodicity, which gave the accuracy of repetitions.
In my configuration, the Z axis has 3200 steps/mm, and the extruder has 196 steps/mm. Perhaps because of the large number of steps, they were lost with some periodicity. It is a pity I do not have an oscilloscope, look at the uniformity of the intervals between the pulses of steps.
Given all the above, I have a few questions.
1. Should I change other driver control settings? If necessary, what values should be set? I mean
```
# define MINIMUM_STEPPER_DIR_DELAY 650
# define MAXIMUM_STEPPER_RATE 250000
```
2. If I set the value of #define MINIMUM_STEPPER_PULSE **3** then this also applies to the tmс2208 drivers. How it will affect the maximum speed of moving of XY axes. Same thing with the other parameters from the first question, if they are installed. _I guess that I will don’t feel it, but I still want to get a confirmation._

DimchikUA, 2019-04-24:
The problem of skipping steps did not disappear `#define MINIMUM_STEPPER_PULSE 3`. Just became less noticeable. I had set `#define MINIMUM_STEPPER_PULSE 5`. Looks all good, for now. Bought lv8729 for Z and У axes, waiting for delivery.
Interestingly, there were no loss of steps when I used DRV8825 on all axes and #define MINIMUM_STEPPER_PULSE 2. When I inserted TMC2208 for XY axes, omissions of steps of the Z axis began. There were no omissions on the E axis. The axis Z - 3200 steps / mm, the axis E - 196 steps / mm.

closed 2019-05-01 by DimchikUA