MarlinFirmware/Marlin #13786: [FR] Include Alfawise Ux0 boards in next STM32 boards supported

hobiseven, 2019-04-22:
Hello,
We completed the code port of Marlin 2.0.X running on Alfawise, and it would be great to include this family of boards in the next STM32 port. All coding has been using @JMZ52 great code. 
Our git is now public, based on a pull from Marlin 2.0.X on 14/04/2019. 

Major differences are : 

1/ the touchscreen code, which is close, but not same as MKS robin, due to software SPI needed, as well as compatibility more for XPT2046 chip ( using penINT) as Alfawise uses ASD7843 which is a subset of XPT2046. 

2/ LCD detection is using a different register than the one from MKS_robin. Chinese LCDs do not always rely on ID register 0x04. Our need to be read in 0xD3

3/ LCD type is ILI9341, not supported in MKS Robin. That was added.

4/ Open collector pins are needed on bed, extruder and fan outputs.

5/ Timer 2 clash with PA15 fan pin assignement. As a result the soft PWM for the fan is currently used. 

6/ Touch screen calibration could be included in main code. So far we use JMZ52 tweaked code to get calibration data externally. thresholds are not yet right, and it is difficult to get the crosses ticked, as we have a bit of variation in the touch panel resistance, as well as 2 types of touch screens, with inverted Y in one of the types. 

Testing is now ongoing, ad we have 10+ people running / testing, and this is the first day! 

repository : 

https://github.com/hobiseven/Marlin_2.0.X_Beta_Alfawise_Ux0

Thank you for the awsome work!

sjasonsmith, 2020-11-10:
We have Alfawise U20 examples, which use the LONGER3D_LK board. Is this what this request refers to?

thisiskeithb, 2021-09-24:
Support was added with `BOARD_LONGER3D_LK `:

https://github.com/MarlinFirmware/Marlin/blob/35ad3b0d3b1a8bbf6b220f30f63fd0e32be8fe34/Marlin/src/core/boards.h#L334

closed 2021-09-24 by thisiskeithb