MarlinFirmware/Marlin #13787: [FR] USB device Serial Number

FelixHauser, 2019-04-22:
I have a Raspberry Pi with four instances, each for a different printer. In order for the Pi to differentiate among the printers some USB-device data is necessary. For instance the vendor ID, device ID and so on. When you have more that one identical board, the way to differentiate is the Serial Number, which is different for all devices. 

For 8-bit boards I never had an issue. When looking for the info about the device everything was there.

Recently I tried to get the USB-device info for a BIQU SKR v1.1 and all the info I got back was Marlin related. The SN was 1.00000000

I am assuming this serial number is hard coded somewhere in Marlin.

Would it be possible to include it on configuration.h ?

Thanks in advance

gloomyandy, 2019-04-22:
It is slightly harder than you may think. The USB code is not actually part of Marlin, it is part of the LPC176x framework, so it will need changes to both Marlin and the framework to make a change to the serial number data. @p3p what do you think? What's the best way to do something like this? Maybe add a parameter to the USB_init function? Also note that this will have to be implemented for each of the HALs that support USB functionality.

p3p, 2019-04-22:
Having an api to change some of the usb constants is something we do need in the framework, it's doubtful it will ever be used for things other than Marlin but if it is the Product/Vendor IDs and other constants will all need changed.

Passing the values into USB_Init and MSC_SD_Init (for the inquiry string) is probably the least invasive way of doing it.

p3p, 2019-04-22:
Although if its just a serial number we can pull that from the LPC176x I think they all have a unique one accessed with IAP.

gloomyandy, 2019-04-22:
Ah now that's a good idea!

p3p, 2019-04-23:
@FelixHauser If you update your PlatformIO frameworks and libraries to get `framework-arduino-lpc176x       @ 0.1.3` you should now have a unique Serial Number for LPC176x USB devices.

FelixHauser, 2019-04-23:
Awesome! I'll try it first thing tomorrow and post a feedback!
Thanks!

doggyfan, 2019-04-24:
I entered "platformio platform update" into a platformio new terminal window and it updated 3 items OK including framework-arduino-lpc176x to said version.

Marlin compiles fine, though not able to try it on the SKR 1.3 board at this moment in time.

FelixHauser, 2019-04-24:
I just updated my PlatformIO and compiled my Kossel's Firmware. On my macOS computer it reads a very long Serial Number.
I have a BIQU SKR v1.1 board, by the way.

I still have a question though, is this serial number that of the board or is it software generated?.

Thanks a lot for addressing this request so promptly!

[image]

p3p, 2019-04-24:
> I still have a question though, is this serial number that of the board or is it software generated?.

It's the LPC176xs Serial Number they are 128bit unique values set by the manufacturer of the chip.

doggyfan, 2019-04-24:
> It's the LPC176xs Serial Number they are 128bit unique values set by the manufacturer of the chip.

Your "ReadDeviceSerialNum" function returns a much smaller 32-bit value, which is probably enough to get a unique number for whatever use. Any plans to provide a full 128-bit returned value ?

FelixHauser, 2019-04-24:
> > I still have a question though, is this serial number that of the board or is it software generated?.
> 
> It's the LPC176xs Serial Number they are 128bit unique values set by the manufacturer of the chip.

Thanks a lot! I´m closing the FR.

closed 2019-04-24 by FelixHauser

p3p, 2019-04-24:
@doggyfan `ReadDeviceSerialNum` takes a pointer to a `uint32_t[4]` as an argument and fills that array with the 4 `uint32_t` values as there is no 128bit type, your code above has a buffer overrun.

This is how its used to read the ascii hex value for the usb device descriptor (the descriptor array uses 16bit chars (split for the buffer), not null terminated, and has 2 bytes of meta data at the start [length, type])
```c++
  uint32_t lpc176x_serial_number[4];
  __disable_irq();
  ReadDeviceSerialNum(lpc176x_serial_number);
  __enable_irq();
  uint8_t index = 0;
  for(uint8_t x = 2; x < 66; x += 2) {
    uint8_t c = (lpc176x_serial_number[index / 8] >> (28 - (4 * (index % 8)))) & 0xF;
    str_descr_serial[x] = (c < 10) ? (c + '0') : (c - 10 + 'A');
    ++index;
  }
```

doggyfan, 2019-04-24:
oops !   My mistake.

Thank you for pointing that out p3p.