MarlinFirmware/Marlin #13792: [Question] - SKR 1.1 - reuse of XMAX, YMAX and ZMAX limit pins as CS for TMC2130

DavidThijs, 2019-04-22:
In the PINS_BIQU_SKR_V1.1.h file is indicated that the endstop pins cannot be reused to serve as a chip select for the TMC2130 driver boards. 

Quote:
// different type of display (like the TFT), using the pins normally used for the display and encoder.
  // Unfortunately, tests show **it's not possible to use endstop and thermistor pins for chip-select**.
End Quote

I'm wondering why since the LPC1768 datasheet doesn't detail any specific reason why P1.28, P1.24 and P1.26 couldn't be used ? In effect, the datasheet lists them amongst other purposes also as a "General purpose digital input/output pin"

The other options that are available don't suit my purpose (eg. no lcd or no sd), so I'm willing to test any other option available - but I do lack the knowledge on the HAL for the LPC1768.

Currently, I'm using a MKS GEN L with both a MKS TFT32, 128x64 graphic lcd, 4x TMC2130 and a BL Touch. I bought the 32 bit board as I feel the 8 bit Mega2560 isn't powerfull enough when enabling scurve, linear advance or other advanced options.

Any explanation would be welcome !

ManuelMcLure, 2019-04-22:
I expect that this is probably because those pins have pullup/pulldown resistors and filtering capacitors added on the SKR 1.1 board and that's interfering with their use as output pins. See the schematic at [https://github.com/bigtreetech/BIGTREETECH-SKR-V1.1/blob/master/hardware/SKR-V1.1SchDoc.pdf](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.1/blob/master/hardware/SKR-V1.1SchDoc.pdf)

gloomyandy, 2019-04-23:
I wrote that comment after extensive testing of the various options. Indeed it is because of the various pullups and other components attached to the pins. I suppose if you are very determined you may be able to identify and remove the various filter components. You could just try it I suppose you may get lucky. 

But by far the best option is that also described in the pins file. Remove the enable line from the driver modules and wire the enable pad on the driver directly to ground (this makes the driver always enabled at the hardware level), this then frees the enable pins for use as chip select. You then define the option to use software enable which uses SPI to enable/disable the drivers. This was the option I used for my 2130s when I was using them (I've since switched to use TMC2208s). If you search the closed issues you should find a post from me that describes things in more detail (along with pictures).

DavidThijs, 2019-04-30:
Andy, thanks a lot for this torough explanation. I indeed found in the schematics that there are some pull-ups etc... used but I'm not going to remove them for testing purposes (altough I have SMD soldering equipment). I'll give it a go in the weekend with the software SPI enable/disable.

Kudos !

closed 2019-04-30 by DavidThijs